pvvx/ATC_MiThermometer #177: Connection stability issues on LYWSD03MMC with 3.5 firmware

steven-ellis, 2022-01-08:
I think I've  tweaked one of the bluetooth settings and now I can't get a stable connection with this device. It was running firmware 3.1 fine but I upgraded to 3.5 in the hope of better power management support with the pvvx mode. 

Other devices I've updated to 3.5 are working ok.

Hardware Revision String: B1.4 

I can occasionally get a connection for a couple of seconds and then the device disconnects. During one of these connects I managed to get the following

Broken sensor

- Custom config: [39, 8, 0, 0, 80, 4, 169, 124, 49, 184, 60]

Working sensor

- Custom config: [37, 0, 0, 0, 40, 4, 169, 124, 49, 184, 60]

In addition sometimes using the standard OTA tool I get the following message

13:33:03: Detected alternative Firmware? Use: https://pvvx.github.io/ATC_MiThermometer/TelinkOTA.html

steven-ellis, 2022-01-08:
My issue looks very similar to the problems under #67

opape, 2022-01-08:
Maybe it is the same problem I had a few days ago.
I have seven B1.4 devices and was able to flash v3.5 on all devices without any problems.
On one device I had only a short connection during configuration. Exactly as you describe it. The time was not even enough to configure a parameter. On another device the connection broke down when saving the configuration.

In both devices the battery had less than 50% capacity. In the first case it was only about 30%. After I inserted a new battery, both devices could be configured without problems.

steven-ellis, 2022-01-08:
@opape thanks for the tip. After changing the battery I've managed to re-flash the device back to stock and then back to the 3.5a firmware. Odd issue

closed 2022-01-08 by steven-ellis