pvvx/ATC_MiThermometer #179: Can the reed switch change cause accelerated advertisement?

penright, 2022-01-10:
Should a change in the reed switch cause an advertisement? If not can it be done? Maybe when a change in reed switch the interval for advertising gets changed to maybe .5 a second for 2 seconds?

The use case is when ATC is modified with the reed switch, it is hard to use it in any kind of automation. For fun, I modified the 'pvvx_mithermometer' component in ESPHome to expose the flag bit. I can see bit 0 toggles as I make and break the connection, but I have to wait for the next advertisement to publish.

pvvx, 2022-01-12:
The polling of the reed sensor occurs only during the processor wake-up cycle for the transmission of advertisements.
There is no current on the reed sensor while the processor is sleeping. The state of the reed switch is only detected by temporarily connecting the internal pull-up resistor in the processor. 
Connecting a 1M resistor permanently will cause a 3.3 μA constant additional current, which will affect battery life. For this reason, the internal resistor is only connected at the time of measurement.
Ultimately, the response time to a change in the state of the reed switch is the total measurement interval time.
The interval between polls is quite long and it makes no sense to create the transmission of high-speed messages about the change in the state of the reed switch.
Converting a thermometer into a door opening sensor requires a specialized creation of an optional version of the program, which is not needed by everyone due to the higher consumption of the battery.

penright, 2022-01-12:
That makes sense. Thanks for all your work on this project.

closed 2022-01-12 by penright

nadnerb33, 2022-12-27:
I have a similar use case. I want to monitor when a gate is open (so I don't actually care about the temperature readings), I just want a battery powered sensor that can trigger events in Home Assistant. I already use these devices for temp/hum readings in each room, so makes sense to re-purpose one.

I am thinking of wiring a NC reed switch between the battery and the device. When the gate is closed, the NC reed switch will be open so the battery is disconnected. When the gate opens the device powers up and sends the standard BLE advertisement which can trigger an event.
When the gate closes the device turns off and after a timeout period of no packets received the gate can be considered closed.
It should also increase the battery life since it will be off most of the time.

pvvx, 2022-12-28:
A long time, starting from version 3.7, the handling of the reed switch has changed - The reed switch operation mode was added: Switch and Count.