pvvx/ATC_MiThermometer #180: Qingping Temp & RH M batery level

mmkrysiak, 2022-01-12:
There is big difference between neighbour reports of battery level.
Is it any chance to fix it?
Please take a look at the report below

[20:32:45][D][pvvx_mithermometer:118]: Got PVVX MiThermometer (5A:58:2D:34:11:88):
[20:32:45][D][pvvx_mithermometer:121]:   Temperature: 19.92 °C
[20:32:45][D][pvvx_mithermometer:124]:   Humidity: 39.03 %
[20:32:45][D][pvvx_mithermometer:127]:   **Battery Level: 82 %**
[20:32:45][D][pvvx_mithermometer:130]:   Battery Voltage: 2.946 V

[20:32:57][D][pvvx_mithermometer:118]: Got PVVX MiThermometer (5A:58:2D:34:11:88):
[20:32:57][D][pvvx_mithermometer:121]:   Temperature: 19.89 °C
[20:32:57][D][pvvx_mithermometer:124]:   Humidity: 39.04 %
[20:32:57][D][pvvx_mithermometer:127]:   **Battery Level: 75 %**
[20:32:57][D][pvvx_mithermometer:130]:   Battery Voltage: 2.879 V

[20:33:02][D][pvvx_mithermometer:118]: Got PVVX MiThermometer (5A:58:2D:34:11:88):
[20:33:02][D][pvvx_mithermometer:121]:   Temperature: 19.89 °C
[20:33:02][D][pvvx_mithermometer:124]:   Humidity: 39.04 %
[20:33:02][D][pvvx_mithermometer:127]:   **Battery Level: 75 %**
[20:33:02][D][pvvx_mithermometer:130]:   Battery Voltage: 2.879 V

[20:33:17][D][pvvx_mithermometer:118]: Got PVVX MiThermometer (5A:58:2D:34:11:88):
[20:33:17][D][pvvx_mithermometer:121]:   Temperature: 19.90 °C
[20:33:17][D][pvvx_mithermometer:124]:   Humidity: 39.05 %
[20:33:17][D][pvvx_mithermometer:127]:   **Battery Level: 83 %**
[20:33:17][D][pvvx_mithermometer:130]:   Battery Voltage: 2.947 V

[20:33:22][D][pvvx_mithermometer:118]: Got PVVX MiThermometer (5A:58:2D:34:11:88):
[20:33:22][D][pvvx_mithermometer:121]:   Temperature: 19.90 °C
[20:33:22][D][pvvx_mithermometer:124]:   Humidity: 39.05 %
[20:33:22][D][pvvx_mithermometer:127]:   Battery Level: 83 %
[20:33:22][D][pvvx_mithermometer:130]:   Battery Voltage: 2.947 V

[20:33:38][D][pvvx_mithermometer:118]: Got PVVX MiThermometer (5A:58:2D:34:11:88):
[20:33:38][D][pvvx_mithermometer:121]:   Temperature: 19.93 °C
[20:33:38][D][pvvx_mithermometer:124]:   Humidity: 39.05 %
[20:33:38][D][pvvx_mithermometer:127]:   Battery Level: 75 %
[20:33:38][D][pvvx_mithermometer:130]:   Battery Voltage: 2.879 V

pvvx, 2022-01-14:
Clean the battery and contacts.

mmkrysiak, 2022-01-14:
I don't think this is a problem. Sensors are brand new, batteries to.  I've found out the issue after firmware flash. Sensors with original firmware  don't report battery status at all. Hardware issue?

pvvx, 2022-01-17:
I don't understand your problem. 
The battery voltage is different and depends on the load. The load at work is different and the voltage is different. In advertising mode, there is active and passive scanning - with active, there is an additional transmission and more consumption. The battery voltage is lower for a while after this event.

pvvx, 2022-01-17:
It is possible to oxidize the GPIO_PB5 pin on the SoC TLSR8253F512ET32 (CGG1)?
In v3.5c, battery voltage measurement is done using GPIO_PB0, which does not come out of the chip package.

mmkrysiak, 2022-01-19:
I'll try to explain. I 've just flashed several brand new xiomi temp sensors and noticed that battery voltage and level for CGG1s (and only for those models) are very unstable. Please take a look at charts below (pure reports without any filtering)

First for the  LYWSD03MMC wchich looks OK for the reference:
[image]
[image]

and for the  CGG1 (Qingping Temp & RH M) - same time and enviroment
[image]
[image]

For both sensors within shorter period of time
[image]
[image]

As you can see difference between neighbours reports for CGG1 are to big and changes to fast to be true. I don't think it could be explained by the load, temperature etc.

pvvx, 2022-01-20:
CGG1 one of all has internal capacitors for power supply and additional switches for power input from the battery. This may have an effect.
Is this observed on the new v3.5c?

> In v3.5c, battery voltage measurement is done using GPIO_PB0, which does not come out of the chip package.

In the previous version, the voltage power value was digitized from the GPIO going to additional elements...

mmkrysiak, 2022-01-20:
it looks the same on the new v3.5c :(
[image]

Protoncek, 2022-01-20:
That's only under 0.1V difference, which is  actually quite excellent for this device. ADC in ESP32 isn't "state of the art", you know. Good, stable ADC's are pretty expensive. ADC inside ESP32 is definitely only of average quality. It relies on internal voltage refefence, which can fluctuate with chip consumption and so you get varying ADC result.

mmkrysiak, 2022-01-20:
0.1V accuraccy on 3V battery it's not an excellent result. It's makes 10% difference on battery level! 
Note that much cheaper LYWSD03MMC is 10x more precise.
[image]

pvvx, 2022-01-21:
> it looks the same on the new v3.5c :( [image]

This is the effect of capacitors and diodes in the power supply circuit.
Battery voltage is polled after transmitter operation. In the pauses between transmissions, the battery restores the maximum voltage and charges the capacitors to the maximum value through leakage diodes. When a load is applied, the voltage drops and this drop depends on the load and the tasks performed by the processor.

pvvx, 2022-01-21:
I don't speak English and it's hard for me to explain to you.
The impulse load in connection mode is less, because transmiting one frame in 2.5 seconds on one channel.
When working in advertising mode, 3 frames are sent on 3 channels. If the external device has active scan mode enabled, then this is a call to send another frame. Those. Immediately 3 or 4 gears, which has a difference in the supply voltage drop.

Launch https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html
Click "Start Tx Measure" and open debug output to Chrome:
[image]

pvvx, 2022-01-21:
CGG1 in the barn. [GraphMemo.html](https://pvvx.github.io/ATC_MiThermometer/GraphMemo.html)
[image]
[image]

pvvx, 2022-01-21:
CGG1 https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html :
Vbat: 2895 mV
Vbat: 2909 mV
Vbat: 2910 mV
Vbat: 2901 mV
Vbat: 2896 mV
Vbat: 2896 mV
Vbat: 2892 mV
Vbat: 2907 mV
Vbat: 2903 mV
Vbat: 2895 mV
Vbat: 2921 mV
Vbat: 2918 mV
Vbat: 2911 mV
Vbat: 2908 mV
Vbat: 2894 mV
Vbat: 2907 mV
Vbat: 2898 mV
Vbat: 2901 mV
Vbat: 2921 mV
Vbat: 2907 mV
Vbat: 2911 mV
Vbat: 2908 mV
Vbat: 2918 mV
Vbat: 2902 mV
Vbat: 2893 mV
Vbat: 2891 mV
Vbat: 2915 mV
Vbat: 2895 mV
Vbat: 2909 mV
Vbat: 2908 mV
Vbat: 2907 mV
Vbat: 2908 mV

pvvx, 2022-01-21:
Internal Power CGG1
Cycle between image changes on E-ink.
[image]

pvvx, 2022-01-21:
> That's only under 0.1V difference, which is actually quite excellent for this device in my opinion. ADC in ESP32 isn't "state of the art", you know. Good, stable ADC's are pretty expensive. ADC inside ESP32 is definitely only of average quality. It relies on internal voltage refefence, which can fluctuate with chip consumption and so you get varying ADC result. Then any external noise can be the "bad" factor... etc...

This is not an ESP8266/ESP32.
TLSR825x has a good and linear ADC with quality over 12 bits.

pvvx, 2022-01-21:
> it looks the same on the new v3.5c :( [image]

The load on the battery for CGG1 and all with E-Ink is greater. It takes a lot of energy to redraw E-Ink. On the graph, the supply voltage drop reaches 0.2 Volts.
If the voltage was measured without redrawing E-Ink, then it will be 0.1..0.2V more. Approximately equal to the voltage of a loaded battery at 3 mA.
And the redrawing of the screen depends on the stability of the measured temperature and humidity. Voltage measurement together with the operation of the E-Inc driver chip will show 0.1..0.2V less. What we see.

Protoncek, 2022-01-21:
>
> 
> This is not an ESP8266/ESP32. TLSR825x has a good and linear ADC with quality over 12 bits. The stability of the reference source for the ADC is also decent there.

Esp32 was just an example…
I worked with 12 and 16 bit ADCs and among others noise is a great pain in the ***. If, say, a designer doesn’t design PCB correctly a device can have unsolvable problems. GND plating is very important, and not making any loops…etc…
Other thing is, like you say, battery. I have quite a number of cheap batteries which with multimeter show appr. 3.2V, but when i put them, say, in BT button, or remote control (433MHz) it doesn’t work at all (not even for a short time!), because voltage immediately drops to almost zero when loaded with bigger current. In devices like this such batteries can work (if they have a capacitor inside), but above mentioned fluctuation can occur. I added two missing capacitors in all my xiaomi’s. One 20uF and one 100nF.
In this cases a quality battery is VERY important. And, these devices, imported from china have the cheapest “still working” batteries inside.

pvvx, 2022-01-21:
The supply voltage has very high noise - the built-in DC-DC in the TLSR825x works, but this does not affect the ADC

mmkrysiak, 2022-01-21:
Viktor, many thanks for the very good explanation. Because it's a hardware issue there's not so much to do left :(
I'll try to pass reports thru any min/max/median filter over some period of time to have more stable results.

pvvx, 2022-01-22:
Can be bypassed by measurements when there is no E-Ink update:
[image]
Then the readings will be unrealistic. Refusal to work will occur already at 2.2V of such measurements.

mmkrysiak, 2022-01-22:
@pvvx it's a very good idea. Battery level results are much more reliable for typical user now.
[image]
but still need to be filtered to get stable results
[image]

I'm just wondering: 
1. voltage report should be real. If someones decide to take a look on them would like to see reliable values
2. battery level should be stable, otherwise looks suspicious
Can you report battery voltage as it is but for the battery level calculations take only samples without E-lnk update additionally averaged in sliding window (5-7 samples)? 
@pvvx What do you think about this idea?

pvvx, 2022-01-23:
CGG1_v35c_bat.bin + Passive BLE Monitor integration (default settings):
[image]
[image]

pvvx, 2022-01-23:
> 1. voltage report should be real. If someones decide to take a look on them would like to see reliable values

Not possible in principle due to the type of battery.
The voltage dependence of the battery type CRxxx is more than 15 mV per degree. Plus dependence on the discharge.
The new CR2032 on LYWSD03MMC is below 2.0V at -41°C.
The device does not have a current meter or a stable load to determine the internal resistance of the battery - the true remaining charge.
Both dependencies lead to very complex calculations and the result is not yet an indicator of the actual battery discharge.

> 2. battery level should be stable, otherwise looks suspicious

But this is reality.

mmkrysiak, 2022-01-23:
Thanks @pvvx. Mine results are not that stable than yours but much better than before. Please insert changes into the general version for CGG1.
[image]

pvvx, 2022-01-23:
It is possible that such a result was obtained due to the bit depth limitation in the format of advertising transmission you have chosen.
The integration calculates the percentage of the battery itself, according to the voltage values.

closed 2022-01-30 by pvvx