pvvx/ATC_MiThermometer #181: OMG issues

tablatronix, 2022-01-16:
I have 4 units, 3 with ATC firmware work fine, 

I put this version (latest) on one and now OMG wont see it.
All it shows in logs is the service data, Do I need to enable some ad flags or something in the firmware?

Did I overlook some instruction ?

OMG is compiled today from dev

tablatronix, 2022-01-16:
I changed advert type to atc, I guess something in omg config needs to be fixed to read custom, not sure.

pvvx, 2022-01-17:
What is "OMG"?

tablatronix, 2022-01-17:
https://github.com/atc1441/ATC_MiThermometer#using-custom-firmware-with-openmqttgateway

Xploder, 2022-01-17:
Yeah I'm also using the ATC advertisement with OpenMQTTGateway and it works fine. Seems to be less problematic with ESPHome and Tasmota BLE aswell compared to the custom one.

tablatronix, 2022-01-17:
Yeah I assumed this lib would by default retain this capability, but this fork seems to use some other advert structure, not sure why I couldn't find the right details

pvvx, 2022-01-18:
> https://github.com/atc1441/ATC_MiThermometer#using-custom-firmware-with-openmqttgateway

This program skips many BLE promotional packages.
Scans in active mode and drains batteries.
Does not know many already known formats of various BLE sensors.
Gives erroneous readings.
[image]
[image]

tablatronix, 2022-01-18:
There are options for sensors only filtering and passive advertising should be the default. Not sure what your point is

pvvx, 2022-01-18:
Where to enable passive scanning mode?
https://docs.openmqttgateway.com/assets/img/OpenMQTTGateway_Wifi_Manager_enter_parameters.b56b5f33.png
[image]

---

#  "tempc":648.16 ?
{"id":"A4:C1:38:xx:xx:xx","name":"ATC_xxxxxx","rssi":-83,"model":"LYWSD03MMC_PVVX",**"tempc":648.16,"tempf":1198.688**,"hum":74.52,"batt":45,"volt":2.613}

[image]

[image]

---
Out of 20 BLE broadcasts, only 4 are received.
Look sniffer and graph screenshot
https://github.com/pvvx/ATC_MiThermometer/issues/181#issuecomment-1015004817

tablatronix, 2022-01-18:
Are you compiling or using binaries?

pvvx, 2022-01-18:
> Are you compiling or using binaries?

I use the option that is offered to ordinary users.

---
https://github.com/1technophile/OpenMQTTGateway/blob/development/main/ZgatewayBLEConnect.ino#L154
```
BLEdata["tempc"] = (float)((pData[0] | (pData[1] << 8)) * 0.01);
```
signed short, no usigned byte!

tablatronix, 2022-01-18:
shrug, been using it for some time, works well, I did notice the unsigned temp.

```
T: Removing servicedatauuid
T: Removing servicedata
N: Send on /BTtoMQTT msg {"id":"A4:C1:38:B5:36:29","name":"ATC_B53629","rssi":-61,"brand":"Xiaomi","model":"LYWSD03MMC","model_id":"LYWSD03MMC_ATC","tempc":21.4,"tempf":70.52,"hum":46,"batt":45,"volt":2.629}
T: jsonPubl - ON
T: [ OMG->MQTT ] topic: ble/OpenMQTTGateway_ESP32_BLE/BTtoMQTT msg: {"id":"A4:C1:38:B5:36:29","name":"ATC_B53629","rssi":-61,"brand":"Xiaomi","model":"LYWSD03MMC","model_id":"LYWSD03MMC_ATC","tempc":21.4,"tempf":70.52,"hum":46,"batt":45,"volt":2.629} 
T: Creating BLE buffer
N: Device detected: ED:2C:37:8E:7B:B2
T: getDeviceByMac ED:2C:37:8E:7B:B2
T: No device found 
T: Creating BLE buffer
N: Device detected: 22:6B:9F:EF:2D:F7
T: getDeviceByMac 22:6B:9F:EF:2D:F7
T: Get services data number: 1
T: Converted service data (27) to 53fd2802c0aaaba1c0654806b7462ee675f5e5a202125ae9c2ac7f
T: Service data: 53fd2802c0aaaba1c0654806b7462ee675f5e5a202125ae9c2ac7f
T: Service data UUID: 0xfef1
T: No device found 
N: Found 21 devices, scan number 34 end
T: Process BLE stack free: 3112
T: macAdr A4:C1:38:ED:A6:ED
T: isDisc 1
T: isWhtL 0
T: isBlkL 0
T: connect 0
T: sensorModel_id LYWSD03MMC_ATC
```
[image]

[image]

Seems to be working ok now that I switched the ad mode, so not really a big deal.

mosquitto_pub -t home/OpenMQTTGateway/commands/MQTTtoBT/config -m '{"bleconnect":false}'
mosquitto_pub -t home/OpenMQTTGateway/commands/MQTTtoBT/config -m '{"minrssi":-80}'
mosquitto_pub -t home/OpenMQTTGateway/commands/MQTTtoBT/config -m '{"onlysensors":true}'

pvvx, 2022-01-18:
> shrug, been using it for some time, works well, I did notice the unsigned temp.

Does not show negative temperatures for many BLE sensors.

Xploder, 2022-01-18:
Can you recommend ESPHome/Tasmota BLE instead?

pvvx, 2022-01-18:
> ESP32 can't receive BLE normally with WiFi working and sub-optimal overloaded software.

ESP32:
* ESP32 can't receive BLE normally with WiFi working and sub-optimal overloaded software.
* Busy CPU gives additional BLE receive gaps.
* Huge unoptimized code doesn't fit in cache and loads slowly from SPI-Flash.
* The outdated manufacturing process of ESP32 chips leads to high current consumption (Consumes more than WiFi-BLE-Zigbee IoT routers).

znanev, 2022-01-18:
@Xploder I use Tasmota on a ESP32 for almost a year and so far I'm happy with it. If you don't mind the higher consumption, occasionally missed readings etc. shortcomings, mentioned by @pvvx above, it is quite a workable solution. At least I'm able to read my sensors in the garden and the shed (which are too far away for my HomeAssistant instance's BLE adapters).

Protoncek, 2022-01-18:
I used to have Tasmota, too. But eventually i went to ESPHome, since it has way more programming capabilities - your imagination (and skills) are pretty much the only limit. The presence of automations is a biiiiig plus for ESPHome.

I did have quite a number of ESP8266 modules as temp/humi sensors and i replaced many of them with these tiny xiaomi's. They don't need power supply, they are tiny, they are already in nice housing... and i certainly don't need fresh temp/humi info every second (or minute). even every 5 minutes is quite ok.
Now i have three ESP32 BT receivers (one in each floor) and they cover all my house.

tablatronix, 2022-01-18:
well as long as you retain this support, ill close this, works fine for me now. Plus you can always run esp32 with external ble. Any suggested beacons to test with ? I have some itags but thats about it.

Most stuff uses anonymous macs anyway so not sure what you need real time scanning for, but like I said my OMG finds 32 devices, it sees pretty much everything in my house

closed 2022-01-18 by tablatronix

tablatronix, 2022-01-18:
FYI, the signed temp should be fixed in OMG 0.9.9 beta

pvvx, 2022-01-20:
> I used to have Tasmota, too. But eventually i went to ESPHome, 

https://www.home-assistant.io/blog/2021/03/18/nabu-casa-has-acquired-esphome/
>> With the acquisition of ESPHome, Nabu Casa now owns the copyright of Otto’s code and the ESPHome organizations on GitHub, Docker, etc. 

---

> FYI, the signed temp should be fixed in OMG 0.9.9 beta

Next version 0.9.9.9.9.9.9 ? :)