pvvx/ATC_MiThermometer #183: LYWSD03MMC no longer sends info after using 3.5c

comet424, 2022-01-27:
so i noticed there is a new version of the atc thermometer.bin
the 3.5c    as  i figured it fix the issue that LYWSD03MMC  should be only sending info  every 10 min to save battery life

but really its updating more often  when i using it with ESP Home and a wroom32  to monitor it in Home Assistant..  i tried multiple times to save  10 min option but it still updates more often sometimes every few seconds to  a few min

so i was usint the atc_thermometer.bin     
i tried the 3.5c  and it uploads  but it now no longer sends info  in esphome it just shows the mac address it broadcasts..  but it will no longer send data.. 
when i downgrade back to atc_ther...bin  it will transmit data..

and dont know which HW version i have it doesnt say on the unit..

but still cant solve the issue it wont only send info every 10 min

PCjabber, 2022-01-30:
Try disconnecting from the flasher web page, then remove & replace the battery in the thermometer. The Bluetooth icon should not be lit on the display. 

If you need to reconnect with the flasher page, you may need to refresh the browser before it will reconnect.

comet424, 2022-01-30:
nope doesnt work..  when i disconnect the phone  and then add the MAC address to ESPhome  in Home Assistant
it will detect the Bluetooth  Mac Address in ESP Home..   but it will not broadcast the Temperature Humidity. battery life..

i have tried pulling battery etc..

on the Flasher Webpage  you can send commands successfully to the device  (thats what it says)    but actually you cant ..
it will not remove the face or battery.. when you send the Save configuration  to Flash  it will not write  STO  like in the atc_thermometer.bin  file..     

with the atc_thermometer.bin file it all works  you can remove the smile battery... the  send data every 10 min doesnt work it seems to send it randomly  between every few seconds to  5 min or so...   even if you press  the 10 min several times and hit save configuration to flash...

ive tried uploading atc_35c  a 2nd time  after doing it once and it weill actually upload  at 10x  slower  so instead of 68 seconds to upload the firmware  or so    it will actually take 5-10m  to get like 10 percent then crap out saying it failed...

but i can always downgrade back to atc_thermometer.bin

i have tried it with 4 of 11 of my  lywsd03mmcs       they all fail not being able to send data to home assistant  the logger of ESPHome shows nothing

also the TEFlasher  cant read the temperature display...  i dont have it hooked up and flashed.. but it gave some error that you wont be able to view the display as your running custom  or something.. just says for the temp/humidity section it writes you wont be able to view it...

but once i downgrade   i can get it to work with home assistant..  but it just wont do every 10 min to save battery life..   it randomly choooses to send updated info through the blue tooth  so i wont even last 6 months on it... i bet    started off at 100%  like jan 13    and some down to 75% or so one is down to 43%

comet424, 2022-01-30:
so running with the atc_thermometer.bin
this is working   with esphome home assistant..  i  going to mod the bedroom sensor one

[image]

comet424, 2022-01-30:
here is where i modded it and stops working for bedroom
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

comet424, 2022-01-30:
so it will only display the found device  .... 55:5A        it will never display the info
now when i downgrade back to atc_thermometer.bin
when i downgrade.. i do not even have to re save the config  it remembers not to keep the smily face or battery on. and keeps it off.. and it resends data to  home assistant

[image]
[image]
[image]
[image]
[image]

pvvx, 2022-01-30:
"only send info every 10 min" - there is no such version.
The maximum advertise transmission interval in the BLE standard is up to 10 seconds to be able to connect to the device.
Periodic advertise is only included in the new Bluetooth 5.3 standard.
With periodic advertising, it is not possible to connect to the device.
The LYWSD03MMC does not have a button to switch to connectivity mode like the XMWSDJ04MMC does.

The ATC1441 version has an advertising interval of 1.875 seconds and is not configurable.
My version has the ability to set the ad interval from 60ms to 10s.
But it is not recommended to set the interval to more than 3 seconds.  On a lot of equipment, you will not be able to connect to a device with interval settings longer than 3 seconds.

pvvx, 2022-01-30:
The measurement interval is configurable. 
Its value is set in advertising interval periods and ranges from 1 to 25.
If the advertising interval is 2.5 seconds and the measurement interval is 4, then the polling sensors will be  2.5 x 4 = 10 seconds.
If the measurement interval is 25 and the advertising interval is 10 seconds, then 250 seconds will elapsed between measurements.
Battery measurement requires ultra-low consumption and does not affect battery life. 
The battery life is mainly affected by the transmission of advertising packages, active scanning by third-party devices, frequent connections to the device.

comet424, 2022-01-30:
but i guess i miss read it.
its 10 min internal..    i thought that ment  it only updates the screen every 10 min and then sends out a ble  signal .  but i guess ble is always connected.. so its not like a wifi signal  that it only sends when there is a change

and not sure  what you mean  not able to connect to the device  with internal settings longer then 3 seconds?

is this mean that for the atc_35c.bin?

is that why it doesnt work with esphome  and the teflasher?     you cant turn off the smile face or change from C to F   all the features are disabled..

so how would i say make esphome/home assistant to like 10 min like the internal advertisement that would save battery life..

comet424, 2022-01-30:
this is my epshome  a section for it  for the esp32
is there something i can add to make the interval longer?  i did try    update_interval: 5m  or 5000s  but it doesnt like that command

```esp32_ble_tracker:

sensor:
#######################################
### Bedroom Temperature Thermometer ###
### Mijia Bluetooth Thermometer 2   ###
#######################################

  - platform: atc_mithermometer
    mac_address: "A4:C1:38:3B:55:5A"    
    temperature:
      name: "Bedroom Temperature (Xiaomi)"
    humidity:
      name: "Bedroom Humidity (Xiaomi)"
    battery_level:
      name: "Bedroom Battery-Level (Xiaomi)"
    battery_voltage:
      name: "Bedroom Battery-Voltage (Xiaomi)"
```

pvvx, 2022-01-30:
In Home Assistant, data update intervals are configurable. Please refer to its documentation.
ESP Home does not support all thermometer variants and various BLE devices and is not considered here.

comet424, 2022-01-30:
ya i looked 
esphome kinda has update thing   ill have to look in it

and esphome home does work with it.  just not with latest version

ATC_35c.bin  doesnt work with LYWSD03MMC   with  teflasher and Esphome and Home assistant
ATC_thermometer.bin_ works LYWSD03MMC with teflasher and esphome and home assistant

i upgraded to atc_35c.bin as it had latest  update for LYWSD03MMC  and any added features  
but as you see in the pics above it breaks  the unit

but i wanted to download the atc_theremometer.bin  as its not on the github   just the atc_35c     i was going to save it to my home assistant folder...  
i only have it on my phone.. but its not letting me download the file off the phone to the computer for some reason

pvvx, 2022-01-30:
> you cant turn off the smile face or change from C to F all the features are disabled..

Yes :
[image]

comet424, 2022-01-30:
well then why doesnt it work for me above??

atc_thermometer.bin    you see the pics you can turn off smile face you can change to C  turn off Battery

atc_35c.bin  you see in the pics above.. you cant turn  off Smile face you cant turn off battery you cant switch to F  
you cant  save configuration to Flash   it says  send configuration  but there is  no   "STO"    on the thermostat screen

and there is no ACC i think it was  when you program the thermostat atc_theremostat.bin   it does ACC and what not it looks like it reboots the thermostat..    

where as   ath atc_35c  does nothing   as you can see from the pics  

wehre i showed

atc_theremostat.bin
then to
atc_35c.bin
then back to
atc_thermostat.bin

to show its not working

maybe ill make a youtube video  so you can see its not working for me.

and these are all 2021 October  Thermostats

comet424, 2022-01-30:
so i made video  my internet is slow  so going t take a bit to upload  
i show  where once i install the atc_35c  the unit doesnt update.. and youll see as its uploading the File  not even 50 percent i guessing the thermostat factory resets itself?  not sure but ill post when its done

pvvx, 2022-01-30:
For this version ATC_Vx.x.bin, TelinkMiFlasher works from this repository:
https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html
For atc1441 version (ATC_Thermometer.bin) from this
https://atc1441.github.io/TelinkFlasher.html

comet424, 2022-01-30:
oh ok..  ill try..  didnt know i needed a different installer.. ill give it a try
and whats better then 
atc_thermometer    or  atc_35c

as i did the atc_thermometer.bin  from the youtube videos
but when i lost the link  i googled..  and found that atc_35c and then i used the same webpage installer i found..

but ill give it a try thanks

comet424, 2022-01-30:
so i trying  is it normal to be slow like i mentioned i found
1% takes 16 seconds 
so 10%  took 92 seconds

took a bit to figure it out

comet424, 2022-01-30:
so i got it to upload   this atc_35c  
the features on the page.. are they all for this device?  or multiples..

showing time on this doesnt work so well..    but least its working i think..

comet424, 2022-01-30:
so guess i need to use the atc_thermometer.bin file  
even with the other website  to use the atc_35c.bin

the atc_35c  is incompatible with esphome...    and atc_thermometer.bin  is compatible with esphome
so that sucks as i liked what i read that the 35c  saves more battery power
but it just cant see the internals
```
[12:50:47][D][esp32_ble_tracker:720]: Found device A4:C1:38:3B:55:5A RSSI=-86
[12:50:47][D][esp32_ble_tracker:741]:   Address Type: PUBLIC
[12:50:47][D][esp32_ble_tracker:743]:   Name: 'ATC_3B555A'
[12:51:06][D][atc_mithermometer:112]: Got ATC MiThermometer (A4:C1:38:98:BA:1D):
[12:51:06][D][atc_mithermometer:115]:   Temperature: -6.2 °C
[12:51:06][D][atc_mithermometer:118]:   Humidity: 44 %
[12:51:07][D][atc_mithermometer:121]:   Battery Level: 54 %
[12:51:07][D][atc_mithermometer:124]:   Battery Voltage: 2.708 V
```

stage one uses  atc_thermometer..  and the 55 5a  uses the atc_35c

and i forget from  up above i the atc_35c  can send the internal info or not

sorry if i dont make sense or run sentences etc i have dislexia learning disability... it sounds ok to me when i read and write it.. but it may not always sound right to someone else

PCjabber, 2022-01-30:
> the atc_35c is incompatible with esphome... and atc_thermometer.bin is compatible with esphome so that sucks as i liked what i read that the 35c saves more battery power but it just cant see the [...]

if you want to use the atc_35c firmware (from **pvvx/ATC_MiThermometer** repository), then change your ESPHome config from
`- platform: atc_mithermometer`
to
`- platform: pvvx_mithermometer`

So, using your config from your post a few hours ago, your sensor section should look like this:
`sensor:
#######################################
### Bedroom Temperature Thermometer ###
### Mijia Bluetooth Thermometer 2   ###
#######################################

  - platform: pvvx_mithermometer
    mac_address: "A4:C1:38:3B:55:5A"    
    temperature:
      name: "Bedroom Temperature (Xiaomi)"
    humidity:
      name: "Bedroom Humidity (Xiaomi)"
    battery_level:
      name: "Bedroom Battery-Level (Xiaomi)"
    battery_voltage:
      name: "Bedroom Battery-Voltage (Xiaomi)"`

Update your ESPhome, and make sure you are not "connected" to the Xiaomi thermometer via the "flasher" web page (Bluetooth icon should not be visible on thermometer). Within a couple minutes, you should see similar lines in the log but with "pvvx" referenced instead of "atc", and HA should show all 4 sensor details.

Hope this helps!

comet424, 2022-01-30:
ya that helped i had to do it a couple times..  from the one screen as it wasnt populating the  2nd upload screen 
it got stuck  where it says OTEFlasher  or whatrever it is as i dont have the phone on me.. i ended up rebooting phone  and the thermometer.. and i found it has a vB1.7

it was flashing  between F and C  also  after 2 more flashes  it stuck C

but its working..    but i havent touched any of the settings  wanted to get it working to see alternate options then the other one...

 with my dislixia  i didnt even notice the pvvx_mithermometer  and atc one   i found the pvxx link off the esphome 
stupid dislexia  you see words that arent there... so you have to re read things like 20 times sometimes

i hoping this version  can save more battery power.. i started off at 100%  with the stock batteries on jan 13  already they are down to 75 % area  1 is 56%  and one is 43%

as you almost have to change the batteries ever month   rate its going.. but then thats the stock batteries  not sure if its other batteries..

but ya if you figure the best settings be good too.. cuz right now the one video i watched guy said battery lasts over a year
https://www.youtube.com/watch?v=K-HG7qs9hK0

is the guy i learned these theremometers from... but he talks lasts a year  i be lucky to last till super bowl lol

pvvx, 2022-01-30:
> so i trying is it normal to be slow like i mentioned i found 1% takes 16 seconds so 10% took 92 seconds

```
01:56:39: Searching for devices
01:56:56: Connecting to: ATC_0B5EED
01:57:18: Hardware Revision String: B1.4
01:57:18: Detected custom Firmware
01:57:18: Hardware Version: LYWSD03MMC B1.4, Software Version: 3.5, Sensor: SHTC3
01:57:18: Custom config: [5, 0, 0, 0, 40, 4, 169, 124, 49, 184, 60]
01:57:24: File size: 63716 bytes
01:57:24: Count: 3983
01:57:25: Start DFU
01:58:05: Update done after 38.914 seconds
01:58:15: Disconnected.
```

comet424, 2022-01-30:
guess that answers that mine took 5800 ish seconds 

when i rebooted everything and re tried  i took less...

closed 2022-02-02 by pvvx

comet424, 2022-02-05:
ah ok  ya i meant was about battery is.. ya they all work
but what i ment was  i hoping the custom firmware 
will make the LYWSD03MMC  last say 365 days  not   say  60 days as thats what the  battery lifes seem to be reporting  in a way as it hasnt been a month in operation  and they are in below 60%  and the one is 39%  and it started at 100%  

so thats why i hoping the  custom firmware and maybe a good battery.. ill get a year...  at least    not change them every couple months is what i ment about  lasting

and not sure what 10.2020  12.2020  means is that the custom version?

pvvx, 2022-02-05:
> and not sure what 10.2020 12.2020 means is that the custom version?

The time since the first files in this repo is over 14-15 months.
ATС1441 published the first version before September 1, 2020 (this is the [date of his video](https://www.youtube.com/watch?v=NXKzFG61lNs)). 
Since that time, I have been torturing several Xiaomi LYWSD03MMC with constant firmware and checks ... But the batteries in them are still native.

comet424, 2022-02-08:
ah ok thanks.. least i should get some life..  as i display the battery health in home assistant .. and it feels they wont last at at 
the one that was 43%  today is 56%  so not sure why the percentages change   some back  in 76% and 60% and 50%  for 10 units i have and only put them on  Jan 13...  maybe bad batteries  etc.. but hopefully they last 9 months... 

least this custom code is working not sure how much better is from the orginal.. i have  5 and 5  5 running 35v and 5 running thermometer.bin  so time will tell...    least i got my head around things