pvvx/ATC_MiThermometer #185: Flashed the 3.5c and since then the tool cannot connect or retrieve config

OscarHanzely, 2022-01-27:
Using Chrome on linux, with the experimental feature turned on.
When I first connected to the temp sensor everything was as advertised. The Readings were up and connection was successful, the bind keys etc were discovered.

It was showing Hardware Revision String: B2.0 which makes sense its fresh shipment from AliExpress

I did not however see the button to flush the firmware and that probably should have been a warning sign ?

I manually selected the latest bin file of 35c firmware and flash it on. It went all well and finished successfully. 

Since then the device changed the name as advertised. However, now I cannot connect to the device at all. It is completely blocked from connecting. Firmware is also incompatible with B2.0 because its no longer shows correct hardware revision. It usually cannot connect 5 times and then shows that could not connect due to unknown reason.

If I restarted browser got out the battery on device etc. it managed to connect but showed its using some wrong custom firmware and request to get config never returned any data.

```
12:13:00: Searching for devices
12:13:05: Connecting to: ATC_9DEBE4
12:13:14: NetworkError: Connection failed for unknown reason.
12:13:14: Reconnect 1 from 5
12:13:22: Hardware Revision String: B1.7
12:13:22: Detected custom Firmware
12:13:22: Detected alternative Firmware? Use: https://pvvx.github.io/ATC_MiThermometer/TelinkOTA.html
12:13:47: Settings 55 was send successful
```
I see the Bluetooth sign on the thermometer so communication is trying to connect.

OscarHanzely, 2022-01-27:
So I managed to use that TelinkOTA site flusher, no luck connecting so then I took the stock firmware from he site and flushed it there. 

Now its even worse. It reverted back to stock firmware name but immediately both flashers show just errors

```
12:57:34: Connecting to: LYWSD03MMC
12:57:34: NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
12:57:34: Reconnect 1 from 5
12:57:34: NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
12:57:34: Reconnect 2 from 5
12:57:34: NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
12:57:34: Reconnect 3 from 5
12:57:34: NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
12:57:34: Reconnect 4 from 5
12:57:34: NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
12:57:34: Reconnect 5 from 5
12:57:34: NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
12:57:34: Something went wrong, to many reconnect's
```

bcc, 2023-02-19:
I was seeing the same issue using Chromium on Linux (ubuntu 22.10 currently) - what worked for me was to flash them, then remove the pairing for the devices in bluetooth manager, then from a terminal, run sudo systemctl restart bluetooth - I was then able to re-pair with them and configure as normal. I assume something in the linux BT stack is unhappy about the device name change.

iandobbie, 2024-01-18:
I see the same on a Macbook. Once you have connected once the system locks up and wont talk properly to the remote device. Turning off bluetooth for a minute or 2 resets things and it works for the next connection.

closed 2025-04-23 by pvvx