pvvx/ATC_MiThermometer #186: Unable to build the firmware in Linux

unaiur, 2022-02-01:
I get this error
```
make: ./src/../utils/tl_check_fw2.exe: Permission denied
```

It seems that you are using a Windows EXE in the final steps of the build process that cannot be executed in Linux.
The original ATC_MiThermometer is not using it, so I guess is optional.

Is there any side-effect if I replace  ./src/../utils/tl_check_fw2.exe with an empty shell script?

unaiur, 2022-02-01:
I get some warnings if I skip that tool:
```
Warning: Undefined symbol 'hci_rx_fifo'!
Warning: Undefined symbol 'hci_tx_fifo'!
```

Are they normal? Can I proceed and flash the firmware I just build?

unaiur, 2022-02-01:
Ohh, I found it: it generates the CRC of the firmware and cannot be flashed without it.
Parent project uses a python script instead to generate the CRC, which is cross-platform and is included in the SDK:

```
       python3 $(TEL_PATH)/make/tl_firmware_tools.py add_crc $(BIN_FILE)
```

Is there any reason not to use it? Will you accept a pull request changing it back?

pvvx, 2022-02-02:
This version gives a different CRC. It is not suitable for OTAs created on new SDKs and does not work with Mijia firmware.

unaiur, 2022-02-02:
Oh, I understand now why I was not able of upgrading the firmware. Do you
where the source code of that exec file is? I would like to compile it in
Linux.

pvvx, 2022-02-03:
I don't have the source code for "tl_check_fw2.exe".

pvvx, 2022-02-03:
Found 'check_fw' from
https://android.googlesource.com/platform/hardware/telink/atv/refDesignRcu/+/refs/heads/master/readme

copied to https://github.com/pvvx/ATC_MiThermometer/blob/master/utils/check_fw

unaiur, 2022-02-03:
After playing a bit with the check_fw command, I think that both versions can be replaced by this simple python script:
```
#! /usr/bin/env python3
import binascii
import sys

with open(sys.argv[1], 'rb') as f:
    crc = binascii.crc32(f.read(-1)) ^ 0xffffffff

with open(sys.argv[1], 'ab') as f:
    f.write(crc.to_bytes(4, byteorder='little'))
```

pvvx, 2022-02-04:
> After playing a bit with the check_fw command, I think that both versions can be replaced by this simple python script:

That's not all. It is required to check the size of the multiplicity of 16 bytes in the header and the CRC must be concentrated at the end of the binary at an address divisible by 16.

unaiur, 2022-02-04:
Let me recap:
- It must add 0s until file length is multiple of 16
- It must patch the file size inside the binary, using elf tools to locate the position
- Finally, we calculate the CRC and add it to the file.

Is that all?

pvvx, 2022-02-04:
It's better to add 0xff up to a multiple of 16 bytes, and then a checksum of 4 bytes.
The size in uint32 [+0x18] should be 0xXXX4.

SDK from Ai-Thinker (build from ATC1441) is not correct for OTA of most devices on TLSR825x.

pvvx, 2022-02-04:
Added OTA firmware test (size, head, head size, CRC) to TelinkMiFlasher.html.
Now it does not flash ATC_Thermometer.bin from ATC1441.

unaiur, 2022-02-04:
Also, the check_fw modifies 2 bytes at offset 6 from 0x0000 to 0x5d02. According to the source code, in that place there is an asm instruction "tj	__reset". Is it important to also change that address?

unaiur, 2022-02-04:
In any case, if you are interested in replacing that binary blob for some source code:

```
#! /usr/bin/env python3
import binascii
import sys

with open(sys.argv[1], 'rb') as f:
    firmware = bytearray(f.read(-1))

# Ensure FW size is multiple of 16
padding = 16 - len(firmware) % 16
if padding < 16:
    firmware += b'\xFF' * padding

# Fix FW length
firmware[0x18:0x1c] = (len(firmware)+4).to_bytes(4, byteorder='little')

# Add magic constant
firmware[6:8] = b'\x5d\x02'

# Add CRC
crc = binascii.crc32(firmware) ^ 0xffffffff
firmware += crc.to_bytes(4, byteorder='little')

# Write the new firwmare back to the file
with open(sys.argv[1], 'wb') as f:
    f.write(firmware)

print(f"Firmware patched!!!")
```

pvvx, 2022-02-05:
> Also, check_fw modifies 2 bytes at offset 6 from 0x0000 to 0x5d02. According to the source code, in that place there is an asm instruction "tj __reset". Is it important to also change that address?

offset 6 - nothing there. Maybe it's a version number or a flag.

closed 2022-05-01 by pvvx