pvvx/ATC_MiThermometer #187: How to send display data

unaiur, 2022-02-01:
I have modified your firmware to support CGDK2 device, but I do not know yet the how the LCD memory is mapped to the segments in the LCD screen.

I was planning to use CMD_ID_LCD_DUMP to send different bit-patterns and identify each of the segments, but I cannot find any tool to send that command.

Is there any HTML page already prepared or should I modify some of the other pages?

pvvx, 2022-02-02:
I usually insert a procedure that automatically moves the bit in the buffer after a couple of seconds and shoot the indicator on the camera.

closed 2022-02-04 by unaiur

pvvx, 2022-02-10:
DIY: CGDK2-2 + TB-04 (TLSR8253)
https://pvvx.github.io/CGDK2/CGDK2-2/
Added to main code and TelinkMiFlasher.html :)