pvvx/ATC_MiThermometer #188: [Feature request] Support for XMWSDJ04MMC

kabili207, 2022-02-04:
Would it be possible to support the recently released XMWSDJ04MMC temp/humidity sensor? I picked a few up from https://www.aliexpress.com/item/1005003340389190.html

[image]

kabili207, 2022-02-04:
That's excellent to hear!

pvvx, 2022-02-04:
I can't do all this alone. Or it will take a very long time.
Need help creating MI-OTA javascript code.
This requires full reverse Mi-Home plugin or mijia access/registration. But there is no time for that yet.

derpeter, 2025-11-07:
Hello, from what i understand the this device should be now supported. While trying to connect with OTA web app i run into 

> 9:55:52 AM: Not found Telink OTA service!
> 9:55:52 AM: TypeError: Cannot read properties of undefined (reading 'getCharacteristic')
> 9:55:53 AM: Disconnected.

I tried this with Chrome on Android, experimental features enabled. I would like to know if this is an expected behavior e.g. because of incompatible stock firmware or if i just missed an important  step in the process. 

Thanks for all you work its amazing!

pvvx, 2025-11-07:
The XMWSDJ04MMC uses a different chip than those supported in the projects for this repository...

> The chip manufacturer (Silicon Labs) and Mijia uses a cryptographic signature for the OTA firmware.
As a result, they do not want users to modify something.
Firmware is only possible via JTAG, which greatly limits the capabilities of users.