pvvx/ATC_MiThermometer #189: No issue, just a question

metafisica, 2022-02-06:
Hi, firstly (and most importantly) i am truly amazed at you guys Thanks a lot for your work (think i missed your paypal so i could drop a couple of bucks).

The issue is, i'm doing a short script to get data from the advertisements for logging through a Raspberry. But the advertised temperature seems to have only on decimal place and through the flashing webpage (https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html) it seems you get 2. I know its not important as the precision of the device is surely not on the hundredth of a degree, but it is bothering me.

As i said Thanks a lot, you are truly amazing.

Besides the couple of bucks through paypal, if you happen to come to madrid and need anything, letme know.

pvvx, 2022-02-06:
Advertise format is selected in the settings:

[image]

The 'custom' format has 2 decimal places.
'Atc1441' - 1 decimal place.
'Mi' - temperature has 1 decimal place, humidity only in percent percent.

metafisica, 2022-02-07:
Man, first and above all thanks a lot for your fast and detailed response.

i really appreciate it.

closed 2022-02-07 by metafisica