pvvx/ATC_MiThermometer #190: Unreliable firmware OTA uploads

unaiur, 2022-02-07:
Firmware uploads are quite unreliable. Some times it takes two minutes, other times it takes two hours and most of the time are interrupted.

Is there any setting that can make OTA uploads more reliable?

pvvx, 2022-02-08:
?
```
01:56:39: Searching for devices
01:56:56: Connecting to: ATC_0B5EED
01:57:18: Hardware Revision String: B1.4
01:57:18: Detected custom Firmware
01:57:18: Hardware Version: LYWSD03MMC B1.4, Software Version: 3.5, Sensor: SHTC3
01:57:18: Custom config: [5, 0, 0, 0, 40, 4, 169, 124, 49, 184, 60]
01:57:24: File size: 63716 bytes
01:57:24: Count: 3983
01:57:25: Start DFU
01:58:05: Update done after 38.914 seconds
01:58:15: Disconnected.
```

pvvx, 2022-02-08:
I took for the test "Redmi Note 7" (release date 2019):
[image]

pvvx, 2022-02-09:
This may mean that an adapter with a BT 4 chip is used. It has a low transmitted power and a limited reception range, designed for a working distance of a couple of meters. Changes in the BT 5 standard provide for an increase in the transmitted power and dynamic range of the receiver, which allows to significantly increase the working distance.

unaiur, 2022-02-10:
Okay, it seems that connection is super slow, I got impatient and retried. After retrying it became impossible to connect again. I had to remove the battery, wait ~5 minutes and try again.

Now, I wait until the device connects. sometimes it can take up to a minute, but it almost always connect.

closed 2022-02-10 by unaiur

pvvx, 2022-02-10:
The thermometer does not have its own source to keep working for more than a couple of ms. Accordingly, the problem is in the BT adapter or its driver.

unaiur, 2022-02-11:
This thermometer has a large capacitor… it keeps working for 5-6 seconds
after removing the battery. But of course, the problem are somewhere else,
most likely in the drivers.