pvvx/ATC_MiThermometer #197: Device known ID corrupted.

ezhische, 2022-02-20:
Device known ID corrupted after reflash. How to change to original, to work with mi home?

pvvx, 2022-02-21:
MiHome does not require a device ID to work. The original firmware automatically detects the device type.

pvvx, 2022-02-21:
A. Write original firmware on LYWSD03MMC.
B. Using [TelinkMiFlasher](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html) or [TelinkOTA](https://pvvx.github.io/ATC_MiThermometer/TelinkOTA.html) download '[BLE-Hacker.bin](https://github.com/pvvx/TLSR825x_OTA_Flash_Hacker/raw/main/BLE-Hacker.bin)'
C. Using [TLSR825xOTA_Hacker](https://pvvx.github.io/CGPR1/TLSR825xOTA_Hacker.html) connect to the appeared device "BLE_xxxxxx"
1. Input Flash Addres: 55000
2. Erase Sector 55000 
3. Input data ('B1.4F1.0-CFMK-LB-XXDZ---'): 42312E3446312E302D43464D4B2D4C42 
4. Write data. 
[image]
5. Next Flash Addres: 55010
6. Next data: 2D5858445A2D2D2DFFFFFFFFFFFFFFFF
7. Write data. 
[image]
8. Restore original firmware boot: key "Change Start Boot"
[image]
5. Close "TLSR825xOTA_Hacker"

ezhische, 2022-02-21:
[image]
Have the same error adding to Mihome

pvvx, 2022-02-21:
Это не связано и с "Device known ID corrupted".
Такое сообщение в MiHome бывает и когда сервер не отвечает.
Для восстановления в MiHome потребуется до 4-х попыток. Причина кроется в старых ключах, которые создавались при активации вне MiHome. Перед прошивкой в официальную версию желательно было в меню TelinkMiFlasher.html выбрать:  "Стереть Mi ключи" или заменить новый ключ на старый, если он был создан в MiHome:
[image]

ezhische, 2022-02-21:
> Custom прошивка не поддерживается в MiHome.
> Но возможна работа с xiaomi gateway 3 при регистрации MiHome на официальной прошивке, затем custom прошивкой термометра с восстановлении bindkey от MiHome, заменой имени устройства на "LYWSD03MMC" и установке типа рекламы - "Mi" или "All" с флагом "Encrupted".

Можно ничего не восстанавливать, достаточно залогиниться при прошивки с ключами которые достаешь и не активировать через Telink Flash. После перепрошивки поставить All или MI с флагом "Encrupted".

pvvx, 2022-02-22:
> Можно ничего не восстанавливать, достаточно залогиниться при прошивки с ключами которые достаешь и не активировать через Telink Flash. После перепрошивки поставить All или MI с флагом "Encrupted".

Ранее произведенные проверки показывали, что если не всё соблюдено, через неопределенное время происходит отключение обслуживания на Xiaomi gateway. Возможно требуется изменение и имени устройства, т.к. он периодически включает активную рекламу и опрашивает имена устройств.

ezhische, 2022-02-23:
Я не говорю о предоставлении ключей.
Самый простой и быстрый способ перепрошивки с сохранением поддержки через gateway 3 в HA. 
1. Добавить в MiHome.
2. Через token_extractor.py получить ключи и device id
3. Подключиться через Telink Fasher и ввести ключи и device id(без активациии) нажать login
4. Прошить прошивку custom
5. Снова подключиться и включить режим Mi или All и галку Encripted Beacon 
6. Всё!

После перепрошивки у меня 2 датчика уже более полугода стабильно работают через gateway и шлют обновления раз в 2,5 секунды.

pvvx, 2022-02-23:
Для работы через gateway 3 в HA и MiHome проще прошить в TelinkMiFlasher.html и и нажать кнопку восстановить ключи.
Исключается волокита с token_extractor.py.
Нужен всего bindkey, который можно получить в интеграции gateway 3. Остальные ключи можно удалить.
Только bindkey и MAC участвует в шифровании рекламы. Остальные в custom прошивке не используются.
Есть ещё зависимость от специального 32-х битного счетчика. Если он будет утерян (частые отключения питания или перезагрузки из-за изменения критических опций в TelinkMiFlasher), то gateway тоже может откинуть устройство до синхронизации с оф. прошивкой в MiHome.

---

Там не может быть 2.5 секунды, т.к. при рекламе Mi с шифрованием, цикл передачи всех значений - температуры, влажности и батареи,  происходит за цикл в 16 измерений с их усреднением. Т.е. обновление показаний произойдет за 160 секунд при установках по умолчанию. Так сделано чтобы было меньшее потребление (за цикл всего 3 раза подготавливается зашифрованный блок) и нагрузка на gateway примерно соответствовала оригиналу.
А если включено "All" и шифрование - это немного увеличивает потребление от батареи, т.к. каждую рекламу необходимо зашифровать, что есть доп. нагрузка на слабый процессор и батарею.

closed 2022-05-01 by pvvx