pvvx/ATC_MiThermometer #198: Supporting uptime/battery time in the custom format (0x181a)

csiki2, 2022-02-21:
First, thank you for your library, I am using it for more than a year (since v2.5) and it's perfect!

When I refreshed my new devices, I noticed that for the LYWSD03MMC, I even can set the proper date.
If the device is able to measure the time for this range; is it possible to add extra 2 bytes to the custom format (0x181a) and send the "uptime"/"battery time" in hours (one byte in days might be too small)?
If somebody changes the battery (hard reset, battery voltage significantly greater than the last measurements) the firmware would save the starting date of the battery change somewhere and then just send the date difference in hours in the advertisements.
With this feature we would be able to follow easily the battery life for each device.
Thank you very much!

csiki2, 2022-02-22:
> 1. Changing the advertise format will require third party software changes.
I thought that your code generates the 19 bytes advertisements output and you only need to add the extra 2 bytes. Isn't that the case?
On the reading side everything will work till they didn't expect exactly 19 bytes, but minimum 19 bytes for 0x181a.

> 2. It is not possible to programmatically determine the replacement of the battery and its condition.
https://github.com/pvvx/ATC_MiThermometer/issues/165#issuecomment-995199763. When replacing with a new element with a lower temperature, the voltage in the new element may be less.
Yes, but I don't consider this a big issue or even to be a valid use case.
First. Right, my new battery went to nearly 70% (around 2.8V) at 5-6C, but that is still way higher than a near depleted battery at 36% and 2.5V, room temperature (15+ °C difference!!!).
Second. I expect somebody to replace the battery at same temperature or the new battery in a better temperature (changing outdoor battery). Could somebody mislead the detection? Of course. It's their tool, if they happy with it then do it. The normal users just wont do it, why would they do it?

pvvx, 2022-02-22:
>I thought that your code generates the 19 bytes advertisements output and you only need to add the extra 2 bytes. Isn't that the case?
>On the reading side everything will work till they didn't expect exactly 19 bytes, but minimum 19 bytes for 0x181a.

https://github.com/custom-components/ble_monitor/blob/master/custom_components/ble_monitor/ble_parser/atc.py#L13

csiki2, 2022-02-22:
I see, all of them is 0x181a and it can rely only on the message length?
Totally new format with a version byte as well? :D

pvvx, 2022-02-22:
+ During a cold start, the thermometer does not know the past battery voltage. The voltage of the battery under the influence of the load is different and does not mean anything. At the cold start of the load more.

csiki2, 2022-02-22:
Yes, it would need to save the voltage time to time. Or we would lose that date too?
Does the statistics lost on battery change or at a firmware update?

I would simply say if the battery is above 3V at start and the last save of the battery is below 2.6V then there was a replacement.

csiki2, 2022-02-22:
> I have installed batteries that have been stored for over 10 years. The initial voltage on them was 2.0V. After a day of work, it rose to 2.8V.

Do you consider that a regular usage? :D
Reset the battery start date through the flash tool in such exotic case.

pvvx, 2022-02-22:
And in this exotic case:
If a capacitor is installed in the power supply (these are models from Qingping) and people themselves restore in LYWSD03MMC, then the voltage is always 3.0V. Even with an old battery. This is their property.

csiki2, 2022-02-22:
In the capacitor case you don't get any meaningful battery percentage either.

> Constantly transmitting the battery installation time is a waste of battery life.

Elapsed time, but you are right, 3 bytes already used on battery (I didn't check, but my guess is that the percentage is a formula from battery_mv).
uint16_t    battery_mv;     // mV
uint8_t     battery_level;  // 0..100 %

> Home Assistant allows you to write a script that calculates the time and battery life without resorting to changing the program in the thermometer.

Is it possible to attach persistent values based on - let's say - MAC address and then propagate new properties to the device?

How would adding one extra byte change the battery life?
One advertisement is 14..20 uA at 3.3V for the total 30 bytes?

pvvx, 2022-02-23:
Whole ad package: 10 + adv_data_len + 1 
if 19 bytes advertisements data -> 30 bytes, consumption increases with RF TX by (30+x)/30 times   
if 12 bytes advertisements data  -> 23 bytes , consumption increases with RF TX by (23+x)/23 times

pvvx, 2022-02-23:
> In the capacitor case you don't get any meaningful battery percentage either.

In no case will you get a meaningful percentage of battery charge by measuring only the voltage of CR2032.

csiki2, 2022-02-23:
Thank you for the explanation.
So the conclusion is that the battery percentage we get from LYWSD03MMC is not really good for anything and there is no point trying to measure the battery life in the device itself.

pvvx, 2022-02-24:
Not certainly in that way. The analysis can be performed by a person who is familiar with everything described and has battery voltage graphs, but not a stupid processor in the thermometer itself.

closed 2023-02-18 by pvvx