pvvx/ATC_MiThermometer #199: Force SWire pin?

devbis, 2022-02-23:
I've flashed firmware that may initialize pins incorrectly. And flashing with `TLSR825xComFlasher.py` doesn't start. Is it possible to force to switch the sensor to "swire" mode?

At the moment the chip doesn't respond at all:

```
$ python3 TLSR825xComFlasher.py -p /dev/ttyUSB0 wf 0 ATC_Thermometer.bin
=======================================================
TLSR825x Flasher version 00.00.04
-------------------------------------------------------
Open /dev/ttyUSB0, 921600 baud...
Chip sleep? -> Use reset chip (RTS-RST): see option --tact
```

pvvx, 2022-02-24:
1. Power off.
2. run python3 TLSR825xComFlasher.py -t5000 -p /dev/ttyUSB0 wf 0 ATC_Thermometer.bin
3. Power on.

devbis, 2022-02-26:
It worked! thank you for the great instruction!

closed 2022-02-26 by devbis