pvvx/ATC_MiThermometer #200: Thermometer LED display

joedupa, 2022-02-26:
Is there any way to disable the "onboard" LED display for the Mi thermometer devices? I have several sensors that will be buried in the deep freezers and I will be accessing their readings remotely via external ESP controller. The intention is to preserve as much battery energy as possible. Thanks.

pvvx, 2022-02-26:
Help solve these problems:
* Disabling the data refresh transmission command on the LCD does not save power.
* There is no documentation for LCD controllers.
* There is no command to switch the controller to low consumption mode.
* LYWSD03MMC uses more than 4 types of LCD controller chips.

joedupa, 2022-02-27:
Thanks Victor, what is the operating voltage range of the LYWSD03MMC ? The 14335 battery is 3.6V as opposed to 3V CR2032, wouldn't this kill the sensor?

pvvx, 2022-02-27:
[TLSR8251F512ET24](http://wiki.telink-semi.cn/doc/ds/DS_TLSR8251-E_Datasheet%20for%20Telink%20BLE+IEEE802.15.4%20Multi-Standard%20Wireless%20SoC%20TLSR8251.pdf): Supply voltage 1.8 - 3.6V
[SHT4x](https://sensirion.com/products/catalog/): Supply voltage 1.08 - 3.6V

pvvx, 2022-02-27:
USB powered operation (5V):
[image]
I do not recommend repeating, but if desired, it works with a slight increase in the total current consumption ... :)

pvvx, 2022-02-27:
[image]
[image]
[image]
[image]

joedupa, 2022-02-28:
Very informative Victor, thanks again, just ordered a couple of those: https://www.aliexpress.com/item/1005003637299772.html

closed 2023-02-18 by pvvx