pvvx/ATC_MiThermometer #201: Not receiving data from sensor

JGeeforce, 2022-03-07:
First, thanks for the newer fork of 1441. 

I flashed 2 sensors (B2.0) with the 1441 firmware and got them running in HA with the custom advertising. The problem which I am facing is that I set the advertising interval to 1 min and 5 min. But in the logs and HA I see the data is picked up randomly. Sometimes even after one hour, instead of 1 minute.

So I thought let's test this pvvx firmware. Flashing went fine, used the default settings. In de logs with esphome I see the sensor being seen every now and then, but without the items Temp, humidity etc.  What can go wrong? It is a setting or config?

pvvx, 2022-03-07:
https://esphome.io/components/sensor/xiaomi_ble.html#lywsd03mmc

JGeeforce, 2022-03-08:
Is this the correct mapping?

atc1441 - atc_mithermometer
custom - pvvx_mithermometer
mijia - xiaomi_lywsd03mmc
HA BLE - ????

closed 2022-03-09 by pvvx