pvvx/ATC_MiThermometer #202: How to buy the right CGG1?

unaiur, 2022-03-10:
I have tried 4 times to buy the right CGG1 device in AliExpress but I have received 2 CGDK2 and 2 old CGG1 devices based on Nordic chip.

Can anybody share a link where I can buy the right version?

Thanks!

pvvx, 2022-03-12:
Version M. 
Label:
[image]

BLE name "Qingping Temp & RH M"

https://aliexpress.com/item/32990058321.html

unaiur, 2022-03-12:
I ordered from that seller last month an received a 3 year old device with Nordic chip

p5y, 2022-03-24:
Received 2 devices from Cleargrass store this week, both without battery and with Nordic chip.
Waiting on 2 from Mi Homes store, not holding my breath...
Maybe they switched back to Nordic chip to keep hardware for M and H version similar?

pvvx, 2022-03-24:
https://fccid.io/2AQ3F
[image]
https://fccid.io/2AQ3F-CGG1/Internal-Photos/Internal-Photos-3990168

"Qingping Temp RH M" has this certificate. But in reality, others are produced and sold.
https://github.com/pvvx/ATC_MiThermometer/issues/41#issuecomment-812803456

unaiur, 2022-03-25:
I think they are cleaning up old inventory. The units I received 15 days ago from MI Homes were manufactured in 2019.

p5y, 2022-03-25:
Just checked, they are from 2022.1
[image]

p5y, 2022-03-30:
Received 2 more from Aliexpress Mi Homes store, they're the same batch as above. production date 2022.01, no battery, same packaging.
It looks like it's going to be hard to get a Telink equipped version from now on, maybe we can add this to the wiki @pvvx?

joshuakraemer, 2024-03-23:
I've just received three CGG1 thermometers from Aliexpress. They've turned out to be the CGG1-M 2022 variant with a production date of 2024.01. Flashing and configuring the custom firmware has worked fine.

I've bought from the following listing ("3pcs M version"):
https://www.aliexpress.com/item/1005006203489017.html

cryptogeek13, 2024-05-15:
I also bought one from Aliexpress, here is the listing: https://de.aliexpress.com/item/1005006203489017.html

Production date 2024.03 and flashing worked flawlessly. Maybe old stock is cleared now?