pvvx/ATC_MiThermometer #203: Trying to flash a CGG1 Qingping Temp Sensor

greggotcher, 2022-03-11:
Step 4 of the flashing process says:
After connection is established a Do Acivation button appears. Press this button to start the decryption key process.

I never see a Do Acivation button

The screen says:
Status: Get Hardware Revision String...
Temp/Humidity: waiting notify for data after connecting

The log says:
12:08:40 PM: Searching for devices
12:09:07 PM: Connecting to: Qingping Temp & RH M
12:09:16 PM: Hardware Revision String: 0001

It then just sits there.  Any ideas on what I may be doing wrong?

pvvx, 2022-03-12:
Fixed for CGG1 in [TelinkMiFlasher](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html) version 4.2.
Remember to press the button on the device before connecting to CGG1.

greggotcher, 2022-03-13:
That worked.  Thank you.

closed 2022-03-13 by greggotcher