pvvx/ATC_MiThermometer #204: Battery consumption/drain

JGeeforce, 2022-03-13:
I installed your firmware on the 8th of March, also because it could function longer on a battery. Some days before that I installed 2 others with the other 1441 firmware. I use default settings and custom advertising. It is working fine, but the battery went from 100% (brand new out of the box) to 77%. With that speed it will take 4 weeks to replace batteries. So it seems, the pvvx consumes more.

With the 1441, which are running several days longer, are on 84% at the moment. So also fast to my opinion, but one of them is set to advertise every 1 minute. So it seems, the pvvx consumes more.

Why does it use so much battery? Do I need to adjust some settings? Or are the batteries supplied by Xiaomi bad ones?

pvvx, 2022-03-13:
For compatibility with third party software, the [atc1441 battery percentage formula](https://github.com/atc1441/ATC_MiThermometer/blob/master/ATC_Thermometer/battery.c#L79) is used.

For example, [HA custom_components ble_monitor](https://github.com/custom-components/ble_monitor/blob/master/custom_components/ble_monitor/ble_parser/atc.py#L56), this is how voltage is calculated:
```
volt = 2.2 + (3.1 - 2.2) * (batt / 100)
```
Battery voltage from start:
[image]

pvvx, 2022-03-13:
[image]
In this graph, most devices have an operating time of about 1 year.
The smallest value on the graph is the [original LYWSD02](https://pvvx.github.io/LYWSD02/) using 2 CR2032 batteries (7 months).

JGeeforce, 2022-03-13:
Sorry and thank you for your reply. It is clear for  me now.

Protoncek, 2022-03-29:
That would be wrong information. If cell has, say, 2.5V at -20 degrees it's correct that it shows very low percentage, since at that temperature it will die very soon. If it would show, say 80% at that point it would mislead the user thinking that battery is still pretty good (at that temperature), while it wouldn't be true. The circuit works on real voltage, not on compensated one...

But you CAN still use a battery which died outside for inside modules, since it will recover when heated back to room temperature.

pvvx, 2022-07-10:
Readings of LYWSD03MMC installed outdoors with a new battery.
And that is how it works to this day.
[image]

50494554524F, 2023-03-19:
writing here without opening a new 3d... my stats and a question

currently i'm using LYWSD03MMC + v3.1 + atc1441 unencrypted + 10000ms adv + 6x measurement... all other settings are on default or disabled

with these settings i'm hitting 2 years with branded batteries (probably even more) and 13-14 months with cheap batteries

now that winter season is ending i'm going to update to the latest version (successfull tested on a thermometer, same settings) with bt home + encryption

are there any stats on these settings?! still 12+ months?! just to know when i have to care about battery level...

thanks

pvvx, 2023-03-19:
There is a very large dependence on the battery manufacturer and operating conditions. The difference reaches more than 2 times. If we add the storage time of the battery before sale and the absence of elements (capacity for power supply) in the cheap version of the LYWSD03MMC and the new CGDK2, then the difference can reach up to 4 times.

For outdoor work and with default settings, there is only this schedule so far:
https://raw.githubusercontent.com/pvvx/pvvx.github.io/master/ATC_MiThermometer/img/Weather.gif
And there is already a second battery.
The first battery has passed into another thermometer with a capacitor in the power supply used in the home temperature and continues to work.

pvvx, 2023-03-19:
LYWSD03MMC has different LCD controllers. They have different current consumption. Plus, it depends on the type of sensor and the installed ratings of the capacitors in the power supply.

The version is determined by the addresses of the display and sensor on I2C:

HW | LCD I2C addr | SHTxxx I2C addr | Note
-- | -- | -- | --
B1.4 | 0x3C | 0x70 (SHTC3) |  
B1.6 | UART! | 0x44 (SHT4x) |  
B1.7 | 0x3C | 0x44 (SHT4x) | Test original   string HW
B1.9 | 0x3E | 0x44 (SHT4x) |  
B2.0 | 0x3C | 0x44 (SHT4x) | Test original   string HW

Version 4.2 has several changes for less consumption. And uses a different calculation of statistics as a percentage of the battery.

pvvx, 2023-03-19:
The CR2016 is used for faster battery run time determination. But after 6 months they are still working...

pvvx, 2023-05-24:
Version B1.9 uses an outdated LCD controller - BU9792 - Has the highest consumption of other LCD controllers.
The LCD controller in 1.4 seems to be the most economical.
Sensors, since they are rarely polled, do not have a strong difference in consumption.

50494554524F, 2023-05-24:
as you asked i can give you updated stats about batteries... 

B1.4, firmware v3.1, 10s advertising, no encryption, no additional features enabled, inside home: xiaomi battery that cames with the thermometer is still running after 722 days and 27% charge available.... cheap pkcell died after 14 months, i can reuse them in xiaomi window sensors for additional 4-6 months....

thermometers with v4.2 firmware have been running for two months now and the voltage is sitting always around 2.80v...

so we can heat easily a year and no need to care about changing them :D

pvvx, 2023-06-20:
Another test of Xiaomi LYWSD03MMC B1.9 with **CR2025** was completed - a cheap CR2025 battery from a set of 8 batteries for $1  finally died - the SoC of the thermometer went into deep-sleep with the latest readings of the operating voltage in millivolts:
[image]
and the expectation of a 2.2V boost for a fresh start...

[image]

Almost 9 months... 

PS:
[image]

Of these sets, the CR2032 usually works 2 times less than what original Panasonic CR2032

nagyrobi, 2023-06-20:
2 times less means half?

50494554524F, 2023-09-21:
i upgraded to the latest home assistant version to use new features + i installed firmware v4.3. so, i'm using the new bluetooth integration and i can't put it in passive mode because i have a bluetooth bot.

to reduce battery drain i did:
- use ad flags
- advertising interval 10000ms + 6x measure time
- mijia unencrypted format because, i read here, bthome consumes more energy
- all other firmware features disabled

after two months of tests battery drain seems a lot higher compared to the deprecated passive ble monitor... 

is there anything else i can do?! have i missed any of your tips?! could it help going back to firmware v3.1?!

thanks

pvvx, 2023-09-21:
[image]
[image]

https://bluez-cheat-sheet.readthedocs.io/en/latest/ 
```
activate experimental mode for bluetooth service :
$ sudo nano /lib/systemd/system/bluetooth.service

Add --experimental at the end of the following line :
ExecStart=/usr/local/libexec/bluetooth/bluetoothd --experimental
```

bam80, 2023-12-24:
> And I repeat for you personally, since you do not want to [find information on this topic](https://github.com/pvvx/ATC_MiThermometer/issues/165) before posting questions.

This information is scattered around.
Could we put it in a Wiki here maybe, so one don't have to look for it in the future?

brotaxt, 2024-09-09:
After reading all replies, which advertising type should I choose to get minimal battery drain? Mijia?

pvvx, 2024-09-15:
Mijia has more consumption than BTHome.

pvvx, 2024-09-15:
I don't use ESP32 because it works poorly in both BLE and WiFi. Especially since it only has version BT4.2. It's outdated, always glitchy, and consumes a lot.

brotaxt, 2024-09-15:
I'm using sonoff t5 switches which have an esp32 included. Thats why i'm using the esp32 :)

pvvx, 2024-09-15:
On ESP32-C3, BLE advertising reception is significantly (!) better, but with WiFi disabled. [Supports BT5.0+, including LE long Range](https://github.com/pvvx/ADV_BLE2UART).

nicolaerosia, 2025-01-06:
@pvvx ESP32-C6 should be even better.

https://www.espressif.com/sites/default/files/documentation/esp32-c3-wroom-02_datasheet_en.pdf#page26
https://www.espressif.com/sites/default/files/documentation/esp32-c6-wroom-1_wroom-1u_datasheet_en.pdf#page26