pvvx/ATC_MiThermometer #205: Xiaomi Mijia LYWSD03MMC, firmware 3.7 no data on home assistant

nerdguy1338, 2022-03-15:
Title is pretty self explanatory.

I had the sensors on 3.5, they're working perfectly.

I happen to check for updates, and flash 3.7 to a few of them.

Now they wind show up in home assistant, just "unknown" for temp and humidity data.

In the online flasher, the advertising format drop-down no longer has "all" as an option.

pvvx, 2022-03-17:
The "All" option has been replaced with the new "HA-BLE" format.
[Support home brew DIY sensors](https://github.com/custom-components/ble_monitor/issues/548)

nerdguy1338, 2022-03-17:
Ok but "all" was working just fine for me, and "ha-ble" isn't.

I'm using the ble tracker in esphome, with a couple of esp32 boards polling
some sensors.

In the yaml, I'm just using "pvvx_mithermometer"

pvvx, 2022-03-17:
Old versions of HA ble_monitor custom components are available in 3 formats: atc1441, pvvx, mijia.
The new version will have 4 formats. HA-BLE will be added soon.
The current EspHome supports 3 formats: atc1441, pvvx, mijia.

nerdguy1338, 2022-03-17:
Oh ok. So I just upgraded too early. Ok cool. Thanks for the info!

pvvx, 2022-03-17:
A working beta with 'HA_BLE' is available here:
https://github.com/custom-components/ble_monitor/tree/new-HA_BLE

pvvx, 2022-03-17:
> In the yaml, I'm just using "pvvx_mithermometer"
> Oh ok. So I just upgraded too early. 

You need to switch the format to "custom" and everything will continue to work.

comet424, 2022-03-21:
im guessing it was too soon to upgrade to 3.7c?  as no option  in esphome?
i tried to downgrade to 3.5c  and it wouldnt work either..  as  i had that running..  only fix i found was 
to load  up the atc1441  firmware..   i even tried orginal firmware and then to 3.5c  and still didnt work.. guess it saves something in memory that couldnt be erased..  
so for now  running that atc1441..  as i couldnt downgrade and get it back up and running