pvvx/ATC_MiThermometer #206: B1.9 has garbled screen with v3.5 and v3.7 (v3.4 was working)

bjeanes, 2022-03-21:
Photos of my various sensors (all B1.9, all purchased around November last year):

[image]
[image]
[image]
[image]
[image]

If I have it set to show battery, it does alternate various screens, but they are all corrupt/garbled with no recognisable pattern.

The few that I haven't yet upgraded (still on 3.4) are continuing to work fine but I tried downgrading one of the upgraded ones back to 3.4 and it does not fix the screen.

AFAICT, the sensors all continue to emit their readings -- it's only the screen which I have noticed to be malfunctioning. 

No amount of re-setting defaults, re-flashin, or the "Show" feature of the flasher will fix the screen. Occasionally something I did in flasher would correct the screen, but then it would appear re-corrupted moments later (perhaps the next screen update).

I searched the other GH issues but didn't see anything that looked familiar. Let me know what info I can get you that may help debugging this.

bjeanes, 2022-03-21:
Restoring to stock firmware restores the screen to normal functioning

Fubukimaru, 2022-04-01:
I have the same symptoms in one of my four sensors. All with the 3.7 firmware. I bought them 2 months ago.

dgom, 2022-04-01:
I had a similar problem with one of my units. I solved it by tightening the bracket screws under the battery cover which presses the PCB towards the display and thus improved the connection. You can manually press the bracket first to see if it gets better.

bjeanes, 2022-04-03:
Hmm... Those screws are very tight for me already. However, re-seating the battery a few times _does_ make the screen come good, but only for a few minutes. After an update or two the screen, it is messed up again.

Here's a video. I had to clip it a bit to fit under GH's upload limit, so the corruption happened about 2 minutes after I re-seated the battery:

https://user-images.githubusercontent.com/2560/161453531-5815e24a-d458-44dd-82db-539b48e52b95.mov

bjeanes, 2022-04-03:
Hmm @dgom still might be onto something though. If I press the case really hard, the next update will have a clear display again. I've (over-)tightened the screws and will see what effect it has, but I found pressing merely the battery made it appear more stable. Perhaps adding a tiny bit of extra foam to the battery cover to press the battery against the PCB is sufficient?

Curious why newer firmwares suffer from this issue when older/stock don't. Are they writing to the display segments faster or at lower voltage?

scvsoft-javierrotelli, 2022-04-04:
I'm having the same problem. 

> Occasionally something I did in flasher would correct the screen, but then it would appear re-corrupted moments later (perhaps the next screen update).

I saw this happen once, but couldn't get it to happen again. tried applying pressure and tightening the screws without any luck.

with both original and ACT firmwares the displays works alright. also, as far as I tested it, the sensor works and reports the correct data to my Home Assistant instance.

Fubukimaru, 2022-04-04:
> with both original and ACT firmwares the displays works alright. also, as far as I tested it, the sensor works and reports the correct data to my Home Assistant instance.

Same thing here. I have flashed a failing LCD sensor with the ATC firmware and it works without problems. On the other hand, I unscrewed and tightened the screws on other unit that started failing and now it works correctly.

Maybe there is something in this firmware that makes these contacts very sensitive which was not done in the ATC firmware.

jddonovan, 2022-04-05:
Bad display contacts can be an issue, but it wouldn't look like this. I suspect that the timing between sensor reading and screen refresh is different in the newer firmware.

Since the garbled output doesn't recover on its own, I suspect the display driver is not getting enough power. Bad batch of batteries? The voltage can drop really low when transmitting bluetooth if the coin cell has high resistance.

Try dropping the transmit power (lowers peak power) and setting the temp measurement into power save mode (changes timing). Also fresh higher quality cell batteries might do the trick!

bjeanes, 2022-04-06:
I'm getting this with unused cell batteries (though the stock ones the units ship with, so voltage still could have dropped a bit, but they report 100% battery level).

Squeezing the unit _definitely_ affects it, which makes the timing theory seem less plausible, no? Definitely not my area of experience though...

Fubukimaru, 2022-04-06:
> I'm getting this with unused cell batteries (though the stock ones the units ship with, so voltage still could have dropped a bit, but they report 100% battery level).
>
> Squeezing the unit [...]

I am not entirely sure about this. One of my sensors failed again after squeezing the screws. So I would not discard the timing hypothesis either :\.

jddonovan, 2022-04-06:
The display contacts fail easily if the unit is dropped or mishandled. However, I've never had an issue with a display on a unit that was not dropped. As I've experienced it, bad contacts result in segments going off, never anything extra(!). In the photos/video, it looks like the issue is extra lines, not fewer, which suggests corruption.

Also, if there are multiple units showing this behaviour, and only after an update, I would venture to guess it's related to firmware. Firmware downgrade is a full reset of the telink chip afaik. However, in theory, the display driver chip could contain semi-nonvolatile memory. It would perhaps reset after being without power for hour or so, or by bridging the power pins when the battery is off.

My hypothesis on the timing issue relates to power draw: if the screen is refreshed eg. at the same time as sending radio data, the voltage might drop below 2v on the display controller rail. However, it could also be that there is some other signalling going on at the same time, causing interference with the display control unit.

One could also try adding a capacitor to the power rail. This might alleviate power spike issues, maybe interference as well  I've done that on tens of units to get extra 30% capacity out of coin cells where the chip would otherwise brownout.

bjeanes, 2022-04-27:
> if there are multiple units showing this behaviour, and only after an update

I am experiencing this on 6 sensors. One of which I tightened the screws and it is clear _most_ of the time but still garbled. Downgrading to 3.4 (which they previously work) doesn't fix it. Reverting to stock firmware does.

> Try dropping the transmit power (lowers peak power) and setting the temp measurement into power save mode (changes timing). Also fresh higher quality cell batteries might do the trick!

I will try this today.

> One could also try adding a capacitor to the power rail. This might alleviate power spike issues, maybe interference as well. I've added a capacitor to tens of units to get extra 30% capacity out of coin cells where the chip would otherwise brownout.

What size capacitor? Is this just across the battery contacts or somewhere internally?

I have enough units which are experiencing this issue to try different things with each one to narrow down the root cause.

bjeanes, 2022-04-27:
> Try dropping the transmit power (lowers peak power) and setting the temp measurement into power save mode (changes timing). Also fresh higher quality cell batteries might do the trick!

I lowered tx power AND set power save mode. No change. I re-flashed v3.7 _after_ changing settings and still no change. It definitely rendered the screen clear a few times, but as it continued to update it just got more and more garbled.

> The display contacts fail easily if the unit is dropped or mishandled.

I still have about 8-9 unopened ones and tried one of them for first time. Same thing. Unless my entire shipment was dropped, these are in good condition. Stock firmware continues to be fine (haven't tested ATC).

> As I've experienced it, bad contacts result in segments going off, never anything extra(!). In the photos/video, it looks like the issue is extra lines, not fewer, which suggests corruption.

Definitely extra lines.

bjeanes, 2022-04-27:
> Stock firmware continues to be fine (haven't tested ATC).

ATC1441 also exhibits this behaviour on the one sensor I tested with it just now.

These MACs were all pvvx and all having issues and I've just flashed them back to stock firmware:

```
A4:C1:38:0F:0B:98
A4:C1:38:2B:AF:8A
A4:C1:38:45:37:02
A4:C1:38:94:85:E3
A4:C1:38:98:5D:30
A4:C1:38:AA:05:0B
A4:C1:38:AB:2F:FC
A4:C1:38:BC:AB:08
A4:C1:38:DC:A8:77
A4:C1:38:F6:B8:CB
```

This MAC is on pvvx 3.4d and continues to be fine (as were the others before upgrading to 3.7):

```
A4:C1:38:32:A8:3F
```

This singular pvvx 3.7 works after tightening screws, which I tried to no success on two of the others above:

```
A4:C1:38:2A:9B:6E
```

These sensors were not all from a single shipment, but from 3 separate shipments over about 2-3 months (I can't be sure I have samples from all 3 in this mix, but it's highly unlikely they're all from a single shipment.

pvvx, 2022-04-29:
> Hmm @dgom still might be onto something though. If I press the case really hard, the next update will have a clear display again. I've (over-)tightened the screws and will see what effect it has, [...]

Version 1.9 differs from 1.4 in a thinner rubber gasket-contacts for the LCD.
This is another Xiaomi savings :)

bjeanes, 2022-04-29:
> Version 1.9 differs from 1.4 in a thinner rubber gasket-contacts for the LCD.
> This is another Xiaomi savings :)

Yay...

Any idea would would account for the change in reliability between different firmwares? I've tried all the suggestions in this thread but none were reliable enough so am back on stock firmware for now :(

> This singular pvvx 3.7 works after tightening screws, which I tried to no success on two of the others above:
> 
> ```
> A4:C1:38:2A:9B:6E
> ```

I saw this display garbled today too. Perhaps changing humidity has caused it to be insufficient. The screws simply won't tighten more without risking stripping, so I'll likely revert this to Xiaomi firmware too :(

[image]

pvvx, 2022-04-29:
Reducing the supply voltage for hw1.9 indicated that the LCD controller needed to reduce the frequency of the CLK i2c..

Check the version of ATC_v37d.bin. I2C bus frequency reduced by 20%.

bjeanes, 2022-04-29:
> Check the version of ATC_v37d.bin. I2C bus frequency reduced by 20%.

Just tried on the one which has been most stable but was garbled this morning... So far so good!

[image]

```
13:39:24: Searching for devices
13:40:06: Get Advertising MAC (Web Experimental Features!)
13:40:06: Connecting to: LYWSD03MMC
13:40:37: MAC: A4C1382A9B6E, Bat: 48%, Vbat: 2640 mV , Temp: 20.58°C, Humi: 69.45%, Count: 218, Flg: 15
13:41:03: Reconnect
13:41:23: Detected custom Firmware
13:41:23: Hardware Revision String: B1.9
13:41:23: Detected custom Firmware
13:41:23: Detected alternative Firmware? Use: https://pvvx.github.io/ATC_MiThermometer/TelinkOTA.html
13:41:23: Hardware Version: LYWSD03MMC B1.9, Software Version: 3.5, Sensor: SHT4x
13:41:23: Custom config: [5, 0, 0, 0, 40, 4, 169, 124, 49, 56, 60]
13:41:23: Hardware Version: LYWSD03MMC B1.9, Software Version: 3.5, Sensor: SHT4x
13:41:23: Custom config: [5, 0, 0, 0, 40, 4, 169, 124, 49, 56, 60]
13:41:27: Load firmware file 'ATC_v37c.bin'...
13:41:28: File: ATC_v37c.bin
13:41:28: File size: 67236 bytes
13:41:28: Count: 4203
13:41:37: File: ATC_v37d.bin
13:41:37: File size: 67236 bytes
13:41:37: Count: 4203
13:41:39: Start DFU
13:42:14: Update done after 34.462 seconds
13:42:24: Disconnected.
13:42:44: Reconnect
13:42:46: Hardware Revision String: B1.9
13:42:46: Detected custom Firmware
13:42:46: Hardware Version: LYWSD03MMC B1.9, Software Version: 3.7, Sensor: SHT4x
```

I will try a few more around my house shortly and let you know if they all perform well.

closed 2022-05-01 by pvvx

Sukalo0290, 2022-10-06:
Unfortunately i'm still getting this with v3.8

Hardware Version: LYWSD03MMC B1.9, Software Version: 3.8, Sensor: SHT4x

I have had it happen a few times for various sensors and versions, some version 3.7 then upgraded hoping the update would fix but happens again.

bjeanes, 2022-10-06:
I too have also been getting it again, but only as my batteries have gotten low...

Sukalo0290, 2022-10-06:
> I too have also been getting it again, but only as my batteries have gotten low...

Do you want to reopen the issue?

bjeanes, 2022-10-06:
I didn't intend to. I've just been replacing my batteries sooner than I'd like to, but it's manageable.

In any case, I don't have the permissions to re-open it because I didn't close it.

If the maintainer thinks there is further opportunity to improve this issue, that's wonderful. But if not, that's understandable.

Sukalo0290, 2022-10-06:
I'm not sure if it is battery, mines currently at 43% and has been doing for a while.

Yeah shall see if someone reopens or can look into it more.

Thanks

pvvx, 2022-10-08:
Yes, there are some problems with the LCD driver in HW1.9. When the voltage drops, unstable operation is manifested.

martini1992, 2022-11-29:
I am also seeing this behaviour when battery reports 40% and lower. Is there nothing that can be done in software?

pvvx, 2022-11-29:
According to the documentation, the LCD driver chip is not designed for undervoltage. But the manufacturer put such a chip in the device...