pvvx/ATC_MiThermometer #207: Qingping CGDK2 - Missing LCD information in "low" temperatures

suschmania, 2022-03-23:
Hi there,
Since 3 days I am using the QIngping CGDK2 sensors with the Firmware version 3.7.
One sensors work in livingroom, the other one outside in a "wind and rain protected" area.

Over night the sensor outside looses the "power" (?) to display the values on the LCD display. At the beginning they are very less visible and at the next morning nothing is visible anymore. Bringing it inside does not help, the display can be reactivated only when taking out the battery and wait 2-3 minutes.
I read the values via bluetooth and they are still transferred, so internally the device still works.

I think the battery has too less power for the temperature environment? we are talking here about ~ 5°C at night with humidity of 70 - 80 %.
**Can I increase the battery power to the display that it stays active?**

Attached some infos from the sensor:
```
Status: Hardware Version: CGDK2 2.1.0, Software Version: 3.7
Vbat: 2955 mV , Temp: 23.12°C, Humi: 46.78%, ID: 11258, Reed switch counter: 0, flg: 0x05:r1/t0
```
[image]

[image]

[image]

[data.csv](https://github.com/pvvx/ATC_MiThermometer/files/8331312/data.csv)

Any help welcome, thank you :)

mr-sneezy, 2022-03-23:
Not sure if this means much, but I've had one of the old square sensors inside my fridge for at least 9 months on the original battery, and the LCD is quite readable. Perhaps try a new battery before anything else....
[image]

suschmania, 2022-03-23:
Thank you.

Is there a difference in my case when i switch the Option "Sensor in "LowPower mode"?

pvvx, 2022-03-24:
Yes, the LCD driver chip in the CGDK2 has this behavior. When the power is reduced or the temperature is low, the microcircuit loses its initialization ... There is no solution yet.

pvvx, 2022-03-24:
The datasheet for the LCD controller chip BU9792 has a Vdd (power) parameter from 2.5 to 5.5 V, which is not enough for this application. 
https://datasheet4u.com/datasheet-parts/BU9792FUV-datasheet.php?id=705112

suschmania, 2022-03-24:
Thank you for the answer! Today night the sensor kept active, I was able to read the display this morning. Yesterday I changed the lowpower mode (turned it off). 
I thought that I should use this one compared to the other options, as it was the only one that goes down to - 10 C in its range. The other sensors starting from 0.

But these negative values are a bit senseless when the LCD display can't stay active... Anyway :)

pvvx, 2022-03-24:
Since the beginning of December, [LYWSD03MMC has been working outdoors](https://github.com/pvvx/ATC_MiThermometer/issues/165#issuecomment-995199763). The temperature was below -27C.
So far everything is working fine...

suschmania, 2022-03-25:
Today night it happened again. So i think I will go for a LYWSD03MMC

suschmania, 2022-04-03:
After 1 week of using the LYWSD03MMC everything is working how it should. The Qingping is used in the sleeping room now.

closed 2023-02-18 by pvvx