pvvx/ATC_MiThermometer #209: Values on CGG1 doesn't updates

rosiste, 2022-03-28:
Hi, after flashed 3.7 firmware to the new device, I'm able to read values in TelinkMiFlasher web page and ESPHome & Home Assistant. Unfortunately, the display always shows the same values (25.4 C and 34.3%), which do not change. Even the settings changes in TelinkMiFlasher (such as changing C / F units, the battery status) has no change on display. Before uploading the firmware, the thermometer ran fine with the default firmware for several days and displayed the values correctly. Returning the original firmware also did not help. The button also seems to work normally. Any clue?

sangandi, 2022-04-11:
I habe the Same Problem.

ewysong, 2022-04-19:
Identical issue here as well.

ewysong, 2022-04-20:
Even worse, I flashed the original firmware "[Original_OTA_CGG1_v1.0.1_0093.bin](https://github.com/pvvx/ATC_MiThermometer/blob/master/Original_OTA_CGG1_v1.0.1_0093.bin)" back on to it and it appears bricked now. Still no change on screen, and not showing up as a Bluetooth device to connect to...

pvvx, 2022-05-09:
>Maybe this version has a different way to communicate with the e-ink display?

This will be the 5th version of CGG1

[CGG1 variants](https://github.com/pvvx/ATC_MiThermometer/issues/41#issuecomment-812803456) produced by [Qingping.co](https://www.qingping.co/temp-rh-monitor/overview) ([ClearGrass](https://www.cleargrass.com/))
---

**CGG1-old** ver1, Confirmed FCC ID: [2AQ3F-CGG1](https://fccid.io/2AQ3F-CGG1/Internal-Photos/Internal-Photos-3990168)
https://pvvx.github.io/CGG1_old
nRF52810 Chipset. It has a small Flash in the nRF52810 chip, an additional SPI-Flash is used for OTA - Limited in functionality. The minimum SDK code (BLE SoftDevice) from Nordic is 80% Flash. Terrible economy on details. Lack of any recommended elements for stable operation of nRF52 chips from battery. Underestimated RF transmission power.
Cheaper temperature and humidity sensor SHT30 (due to the high cost of NRF52).
Production 2018
Increased cost because a chip from Nordic is used.
Alternative firmware is in development. Will not be in this repository.

**CGG1-?** ver2, Unknown FCC ID.
https://github.com/kelchm/cgg1-thermometer-firmware
nRF52832 Chipset. Terrible economy on details. Lack of any recommended elements for stable operation of nRF52 chips from battery. Underestimated RF transmission power.
Production 2018..2019 ?
Increased cost because a chip from Nordic is used and Mijia.
Has no alternative firmware capabilities.

**CGG1-M** ver3,  has an official fake FCC ID: 2AQ3F-CGG1
https://pvvx.github.io/CGG1
TLSR8253 Chipset. The workmanship is better than previous models. SDK (BLE) code takes up less than 1/10 of the internal Flash. Additional circuitry of additional elements for better and long-term stability from battery. It is possible to increase the RF transmission power.
Improved temperature and humidity sensor SHTV3.
Production 2020+
The cost is much lower (¥ 99.00), because the chip used is not Nordic, but the Chinese Telink.
[Alternative firmware option](https://github.com/pvvx/ATC_MiThermometer), including ZigBee options (in development). There is also an SDK for Apple HomeKit (I don't support it).

**CGG1-H** ver4, Confirmed FCC ID: [2AQ3F-CGG1H](https://fccid.io/2AQ3F-CGG1H/Internal-Photos/Internal-Photos-4723738)
nRF52xxx Chipset. Terrible economy on details. Lack of any recommended elements for stable operation of nRF52 chips from battery. Underestimated RF transmission power.
Production 2020+
Increased cost because a chip from Nordic is used and Apple HomeKit (¥ 169.00).

---

marciogranzotto, 2022-05-10:
More picture from internals:
[image]

[image]

[image]

[image]

masdeseiscaracteres, 2022-05-27:
Identical issue here. Haven't disassembled the device but according to all external annotations (box and device) it is the same as the one in the pictures. 

I'm thinking of getting the original firmware from a new device. Is there a way to dump the firmware OTA?

pvvx, 2022-05-28:
Read Full Flash and restore old firmware:

1. Using [TelinkMiFlasher](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html) or [TelinkOTA](https://pvvx.github.io/ATC_MiThermometer/TelinkOTA.html) download '[BLE-Hacker.bin](https://github.com/pvvx/TLSR825x_OTA_Flash_Hacker/raw/main/BLE-Hacker.bin)'
2. Using [TLSR825xOTA_Hacker](https://pvvx.github.io/CGPR1/TLSR825xOTA_Hacker.html) connect to the appeared device "BLE_xxxxxx"
3. Read the entire Flash into a file: "Read Full Flash" and "Save to File"
5. Restore original firmware boot: key "Change Start Boot"
6. Close "TLSR825xOTA_Hacker"

masdeseiscaracteres, 2022-05-29:
The e-ink display is updating again (with the original firmware)!

Finally I followed a different path, but thanks for the info anyway! 

This is what I did:

1. Download the latest official firmware. I got it by looking at the URL that the [Qingping+ app](https://play.google.com/store/apps/details?id=com.cleargrass.app.air) is using. I will provide more details on my findings in a new issue.
    - [Copy of the downloaded file (rename to get rid of .txt extension)](https://github.com/pvvx/ATC_MiThermometer/files/8793487/Goose_8253_v2_2_1__update_param__2_2_1.bin.txt)
    - [Official download link](https://qingfs.cleargrass.com/firmwares/202201/Goose_8253_v2_2_1__update_param__2_2_1.bin)
 2. Updated the firmware using [TelinkMiFlasher.html](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html).
 3. Take the battery off for a couple of minutes.
 4. Insert the battery back and voilà!

marciogranzotto, 2022-05-30:
Thank you @masdeseiscaracteres !! I managed to revive my unit with that file

pvvx, 2022-06-23:
Finally got the new [CGG1-M variant (2022/01)](https://pvvx.github.io/CGG1_2022/).
When free time, I need to break the case and get a dump from the new E-Ink controller...

marciogranzotto, 2022-07-08:
@pvvx https://github.com/pvvx/ATC_MiThermometer/commit/2a92f1ac511a304bcd2d7b9657e2c3479e4aca43#r78018682

ewysong, 2022-07-11:
Interestingly my units match the pictures posted by @marciogranzotto exactly, but I have never seen the Thread icon active between the bluetooth and battery icons. 

I was able to restore the original firmware finally and then flashed the beta version. It now displays in Celsius only, and clicking the button on the back does not switch to displaying in Fahrenheit. 

Trying to reconnect to it after that initial flash doesn't seem to work:

```
12:47:34 PM: Disconnected.
12:47:34 PM: Searching for devices
12:47:36 PM: Connecting to: CGG_142D82
12:47:36 PM: NotFoundError: No Services matching UUID 0000fe95-0000-1000-8000-00805f9b34fb found in Device.
12:47:36 PM: Reconnect 1 from 5
12:47:36 PM: NotFoundError: No Services matching UUID 0000fe95-0000-1000-8000-00805f9b34fb found in Device.
12:47:36 PM: Reconnect 2 from 5
12:47:36 PM: NotFoundError: No Services matching UUID 0000fe95-0000-1000-8000-00805f9b34fb found in Device.
12:47:36 PM: Reconnect 3 from 5
12:47:36 PM: NotFoundError: No Services matching UUID 0000fe95-0000-1000-8000-00805f9b34fb found in Device.
12:47:36 PM: Reconnect 4 from 5
12:47:36 PM: NotFoundError: No Services matching UUID 0000fe95-0000-1000-8000-00805f9b34fb found in Device.
12:47:36 PM: Reconnect 5 from 5
12:47:36 PM: NotFoundError: No Services matching UUID 0000fe95-0000-1000-8000-00805f9b34fb found in Device.
12:47:36 PM: Something went wrong, too many reconnects
```

pvvx, 2022-07-11:
> 12:47:36 PM: NotFoundError: No Services matching UUID 0000fe95-0000-1000-8000-00805f9b34fb found in Device.

Have you flashed the firmware: 'CGG1M_v37c.bin'?
If yes, then your BT adapter is the problem. Your adapter does not update the attributes of the device, but uses some past MAC-bound ones. You should reboot the adapter or the entire system.

---

In custom fw, the button is used to send a press/release event to Home Assistant and the like. In Home Assistant, an action is assigned to a button, for example: turn on / off the light ...
Select in TelinkMiFlasher.html:
[image]

ewysong, 2022-08-11:
I was able to flash the firmware 'CGG1M_v37c.bin' to it. Even after I disabled/re-enabled the BT adapter, and restarted the entire system, it was unable to connect again. A completely different computer was able to however.

I did set the PinCode in TeLink Flasher on one of the three I have to try to keep them more secure. Now I'm unable to reconnect to that one. I do not get a prompt for a pin code when trying to pair. I've tried on a Ubuntu and Windows computers, and two Android devices. Two of which had never connected to it before. Is there a way to factory reset that one? Should I open a new issue for it?

Also noticed for the 3 options under "Management GPIO_RS (Reed Switch)" you posted above one is called "Couner". Not sure what that is, unless it's a typo for Counter (but not sure what Counter would do?).

pvvx, 2022-08-16:
In "CGG1M_v37c.bin" there is a bug with WDT that affects the work with pincode. Fixed in "CGG1M_v37d.bin".

ewysong, 2022-08-17:
@pvvx Thanks! Is there a way to reconnect to the one I set the PIN code on, or am I permanently locked out of it?

pvvx, 2022-08-17:
Unfortunately, if a pin code was entered, then only through hardware programming.
[The USB-COM adapter writes the firmware in explorer. Web version.](https://github.com/pvvx/ATC_MiThermometer#the-usb-com-adapter-writes-the-firmware-in-explorer-web-version)
[USBCOMFlashTx.html](https://pvvx.github.io/ATC_MiThermometer/USBCOMFlashTx.html)
_Connect only TX-SWS and GND wires._
Pin SWS: 
[image]

pvvx, 2023-02-15:
New "**CGG1T**" out? [FCCID:2AQ3F-CGG1T](https://fccid.io/2AQ3F-CGG1T)

marciogranzotto, 2023-03-01:
> New "**CGG1T**" out? [FCCID:2AQ3F-CGG1T](https://fccid.io/2AQ3F-CGG1T)

Can't quite see what's the CI being used in their internal photos, but looks like "??2840"
[image]

gazsiazasz, 2023-09-23:
Do somebody still have the TLSR8 HomeKit SDK (825x_sdk_homekit) saved? It looks like that telink has removed it from its website :(

pvvx, 2023-09-24:
https://disk.yandex.ru/d/3kBmFHgHLeve4g

Saeverix, 2023-11-06:
Hi guys, thanks for the amazing work so far. What are your expectations about supporting the new T revision: CGG1T?
The above PCB posted by @marciogranzotto looks really similar to the previous revisions.

I am asking because I am about the order that version. The added Thread support is a valuable addition for me, since it seems to use less battery compared to Bluetooth.

swiergot, 2023-12-06:
@pvvx Is CGG1-M 2022 now supported on master?

pvvx, 2023-12-06:
https://github.com/pvvx/ATC_MiThermometer

https://pvvx.github.io/CGG1/

swiergot, 2023-12-06:
Sorry, took a wrong binary...

Nikolay-Ch, 2024-05-05:
I have several CGG1-Ms from different model-years.
Now they all have firmware v2.2.1.
If I flash the latest firmware v47, devices from 2022, then everything works.
If I flash the 2020 version, they stop updating the screen.
At the same time, they continue to send data packets normally and connection with them via Chrome is possible.
As well as flashing back to version v2.2.1.

Nikolay-Ch, 2024-05-08:
Hi,
@pvvx - are there any workarounds? Or will I not be able to run firmware 4.6 on my devices?

pvvx, 2024-05-08:
[CGG1_v47.bin](https://github.com/pvvx/ATC_MiThermometer/blob/master/CGG1_v47.bin) - firmware for the previous hardware version. 
[CGG1M_v47.bin](https://github.com/pvvx/ATC_MiThermometer/blob/master/CGG1M_v47.bin) - firmware for new hardware version released in 2022.

Nikolay-Ch, 2024-05-08:
> [CGG1_v47.bin](https://github.com/pvvx/ATC_MiThermometer/blob/master/CGG1_v47.bin) - firmware for the previous hardware version. [CGG1M_v47.bin](https://github.com/pvvx/ATC_MiThermometer/blob/master/CGG1M_v47.bin) - firmware for new hardware version released in 2022.

Thanks! It is working now... But, I found, that MAC-address of the devices has changed.
Ex: with original firmware MAC is like this: 12:34:56:78:90:AB, but now it is: DF:12:34:56:78:90.
As if the entire address has shifted one byte to the left

pvvx, 2024-05-09:
Qingping's MAC is stored in a different format. When working with the original firmware, no changes occur. If you recorded the firmware from a different version, then there will be confusion.
MAC can be changed in TelinkMiFlasher.