pvvx/ATC_MiThermometer #212: Little help with config to setup sent interval

JGeeforce, 2022-04-10:
I could use a little help with the config. I want to use your pvvx firmware and type to sent the data to my ESP32 board every minute. As I need this to control a fan in the shower. Which parameter do I need to adjust/set?

pvvx, 2022-04-10:
What is the level of RSSI?
At default settings, the data is updated every 10 seconds.
ESP32 doesn't work well with BLE...

Protoncek, 2022-04-10:
I really don't see any point of changing update interval to longer time, since with default times battery lasts around 1 year... you get nothing but problems.

But, as already stated elsewhere: ESP32 works quite nice with tasmota FW, but not as good with ESPHome. I have currently 8 devices on single ESP32 with tasmota FW. With external antenna it works with all devices in my house and around it.

I hope this will change when ESPHome will switch from arduino to esp-idf framework - it's already available for ESP32-C3modules.

pvvx, 2022-04-11:
> I hope this will change when ESPHome will switch from arduino to esp-idf framework - it's already available for ESP32-C3modules.

It is technically impossible to get stability on the same antenna with BLE and WiFi.

JGeeforce, 2022-04-13:
> What is the level of RSSI? At default settings, the data is updated every 10 seconds. ESP32 doesn't work well with BLE... In Home Assistant with "Bluetooth Low Energy Monitor" integration, I have 25 BLE devices and all of them are received normally, without dropouts. The cheapest [USB BT adapter is used.](https://github.com/pvvx/ATC_MiThermometer/issues/156#issuecomment-1061424217)

So, you don't use a ESP32 board at all? But just a BT usb dongle to receive the signals?

pvvx, 2022-04-14:
> So, you don't use a ESP32 board at all? But just a BT usb dongle to receive the signals?

Yes. BT4.0 usb dongle: [CSR8510 A10](https://github.com/pvvx/ATC_MiThermometer/issues/156#issuecomment-1061444290)

nagyrobi, 2022-05-02:
As described here https://github.com/pvvx/ATC_MiThermometer/issues/149#issuecomment-1072981933 very good results can be achieved with ESP32 and ESPHome, the default settings need to be tweaked a bit. Best is of course ESP32 with LAN module, with WiFi disabled.

closed 2023-02-18 by pvvx