pvvx/ATC_MiThermometer #214: diabling Smiley -  setting always shows (   )  - MHO-C401

elbman, 2022-04-25:
I'd like to disable the smiley on my MHO-C401's, but unfortunately the bracket are still there.

I thing setting does not work on my MHO-C401. (8x) Backets are always shown, even when you select a smiley without brackets

Hope you can fix it

elbman, 2022-04-27:
my configuration:

06:30:53: New custom config: [1, 0, 0, 0, 40, 8, 174, 124, 199, 129, 30]
06:30:53: Settings 55010000002808AE7CC7811E was send successful
06:30:54: Hardware Version: MHO-C401 V_14, Software Version: 3.7, Sensor: SHTC3
06:30:54: Custom config: [1, 0, 0, 0, 40, 8, 174, 124, 199, 129, 30]

pvvx, 2022-04-29:
The E-ink display has combined segments:
[image]
https://github.com/znanev/MHO-C401

elbman, 2022-04-29:
👌 I got it, thanks for the hint

closed 2022-04-29 by elbman