pvvx/ATC_MiThermometer #216: Devices disconnect from Telink Flasher after a few seconds

gknops, 2022-05-01:
I just flashed a couple devices: LYWSD03MMC B1.4, Software Version: 3.7, Sensor: SHTC3

3.7d is the firmware.

These devices disconnect fairly quickly after reconnecting. If I am quick enough I can request a confirmation change, but it does not seem to stick:

9:57:36 AM: Hardware Version: LYWSD03MMC B1.4, Software Version: 3.7, Sensor: SHTC3
9:57:36 AM: Custom config: [18, 0, 0, 0, 40, 4, 169, 124, 49, 128, 60]
9:57:43 AM: New custom config: [17, 0, 0, 0, 40, 4, 169, 124, 49, 128, 60]
9:57:43 AM: Settings 55110000002804A97C31803C was send successful
9:57:47 AM: Disconnected.
9:57:52 AM: Reconnect
9:57:52 AM: Hardware Revision String: B1.4
9:57:52 AM: Detected custom Firmware
9:57:53 AM: Hardware Version: LYWSD03MMC B1.4, Software Version: 3.7, Sensor: SHTC3
9:57:53 AM: Custom config: [18, 0, 0, 0, 40, 4, 169, 124, 49, 128, 60]
9:58:06 AM: Disconnected.

Is this a known issue?

gknops, 2022-05-01:
Re-flashing now also fails:
10:01:20 AM: Hardware Version: LYWSD03MMC B1.4, Software Version: 3.7, Sensor: SHTC3
10:01:20 AM: Custom config: [18, 0, 0, 0, 40, 4, 169, 124, 49, 128, 60]
10:01:23 AM: Load firmware file 'ATC_v37d.bin'...
10:01:23 AM: File: ATC_v37d.bin
10:01:23 AM: File size: 67252 bytes
10:01:23 AM: Count: 4204
10:01:24 AM: Start DFU
10:01:30 AM: Update error: NotSupportedError: GATT operation failed for unknown reason.
10:01:30 AM: Disconnected.

pvvx, 2022-05-01:
1. Discharged battery.
2. Bad connection. The thermometer is far away.

gknops, 2022-05-01:
Thanks. Devices sitting right next to the computer. One device did start behaving after switching the battery (even though the previous one read 73%). The other one appears to have issues, every now and then it starts up with the display flashing in random ways, and when it does connect it seems to be conflicted about its sensor:

11:57:04 AM: Hardware Version: LYWSD03MMC B1.4, Software Version: 3.7, Sensor: SHTC3
11:57:04 AM: Custom config: [5, 96, 0, 0, 40, 4, 169, 124, 49, 128, 60]
11:57:10 AM: Settings 56 was send successful
11:57:12 AM: Hardware Version: LYWSD03MMC B1.4, Software Version: 3.7, Sensor: SHT4x
11:57:12 AM: Custom config: [5, 96, 0, 0, 40, 4, 169, 124, 49, 0, 60]

Anyway thanks for your help, and thanks for your work on the firmware!!!

closed 2022-05-01 by gknops