pvvx/ATC_MiThermometer #221: Bluetooth5.0+ PHY (LE 1M/2M/LongRange:500K/125K), CSA1/CSA2 (Channel Selection Algorithm)

Sukalo0290, 2022-05-24:
Can you explain a bit more about this?

Is this implemented on the sensors from version 3.7? Even if the sensors are Bluetooth 4.2? 

Are they now broadcasting on multiple channels so a BT chip capable of CSA can pick up more readings?

Thanks in advance, just trying to understand it.

pvvx, 2022-05-28:
These options relate only to BLE Сonnection. Advertising packages are still broadcast on 3 typical channels, regardless of BT version. In new versions BT, the advertising package may indicate an additional channel, but in this case it is not used.

hcgpalm, 2022-10-14:
Is this a limitation with the hardware/runtime libraries, or would it be possible to make it advertise over, for example, S8 coded PHY?
The, probably obvious, use case beeing able to monitor the device using the BLE 5 long range features which I suppose would greatly increase the current, rather limited, range(?)

pvvx, 2022-10-15:
BLE LongRange (S2: 500K / S8: 125K) in the current version.BLE LongRange has been supported since version 3.6.

hcgpalm, 2022-10-15:
So it _is_ advertising over coded PHY?

pvvx, 2022-10-15:
Advertising is always at 1 Mbps. Only **Extended** Advertising come in other formats.

hcgpalm, 2022-10-15:
Ah, I see, so what you're saying is that there's no standard for (optional) coded advertisements(?) Essentially making BLE long range per se practically useless since a peer always needs to be able to receive the standard mandatory 1M advertisements anyway, in order to negotiate a coded connection or learn about coded "extended" advertisements?

I know that coded PHY is an optional feature not supported by most devices today. The target audience here is obviously those of us that do have a long range capable adapter and need the extended range.

pvvx, 2022-10-16:
> I know that coded PHY is an optional feature not supported by most devices today. The target audience here is obviously those of us that do have a long range capable adapter and need the extended range.

TLSR825x supports ZigBee and has a ZigBee SDK from Telink.
OTA for the BLE variant does not have the ability to download the ZigBee version. ZigBee has a larger firmware code size.
Dual OTA solves ZigBee firmware. The first block is to change the Flash markup for OTA with ZigBee ... The second is the ZigBee code itself.
But I don't want to support the ZigBee version.

hcgpalm, 2022-10-16:
Sorry, but I don't understand how ZigBee is related to this?

pvvx, 2022-10-17:
ZigBee uses PHY 250 kbps.

DienoX, 2023-01-11:
Will the LongRange setting allow me to use the thermometer over longer distances?

pvvx, 2023-01-11:
> Will the LongRange setting allow me to use the thermometer over longer distances?

No, because manufacturers ignore this feature in Bluetooth 5.0 on user devices.

DienoX, 2023-01-11:
According to the data from the nRF Connect Android application, my device, i.e. the Samsung Galaxy S20, supports extended BLE range.
[image]
In this case, will setting a greater range from the thermometer allow me to connect over greater distances?

pvvx, 2023-01-11:
> According to the data from the nRF Connect Android application, my device, i.e. the Samsung Galaxy S20, supports extended BLE range.

It doesn't say anything. 
All modern smartphones are capable of accepting "Extended Advertising" in any PHY format, but typical advertising is only on 1M PHY.
First understand the standards...

pvvx, 2023-01-11:
Almost every cheap smartphone from last year has the following options:
[image]
But that doesn't mean it will accept standard advertisements in different PHY formats.
To connect, you must accept standard advertisements on the 1M PHY. Only after a connection is established on the 1M PHY is it possible to switch the PHY format.
"Extended Advertising" is the transmission of a header on the 1M PHY that indicates on which channel and in what format the data itself will be transmitted. Those. without receiving on the 1M PHY, you will not know where and how to receive the LongRange message. Reception of standard advertising on the main channels in the longRange format is not implemented on most devices. And apparently this was done on purpose so as not to force ZigBee out of the market or for some other corporate purposes...
Even to receive "Extended Advertising" with LongRange in the "nRF Connect" program, you need to switch the options in the settings.

DienoX, 2023-01-11:
Thanks for the clarification. Is there any leaks or indications that bluetooth over longer distances will work better?

pvvx, 2023-01-11:
https://novelbits.io/bluetooth-long-range-how-to-achieve-results-over-1km/

hcgpalm, 2023-01-12:
After digging deeper into the subject, I do NOT agree that advertising on primary channels is limited to 1M PHY. 
BLE 5.0 permits the ADV_EXT_IND PDU to be sent on the primary advertising channels using Coded PHY.
So it _is_ true that it requires the use of extended advertising.

https://novelbits.io/bluetooth-low-energy-advertisements-part-1/

I also agree that it would significantly impact power requirements. Because S=8 requires 8x the radio time for the same volume of data, because of the overhead of extended advertisement, and because it still needs to do legacy advertisement i parallel.

Nevertheless, I'm tempted to try to cook up a proof-of-concept to explore this further, although I'm not sure I will be able to.

DienoX, 2023-01-12:
If you managed to do something, please share it. Personally, I hope that bluetooth in the new versions will enable long range without extended advertising.

pvvx, 2023-01-13:
Experimenting with the inclusion of LongRange for "extended advertised" on the main channels show that none of the available adapters in the application for 'HA' does not accept this option. Also checked dozens of smartphones in nRFConnect released that year - the situation is similar.
"Extended advertised" transmitte with header on 1M PHY and data on S8 is successfully received by most.

pvvx, 2023-01-13:
> If you managed to do something, please share it. Personally, I hope that bluetooth in the new versions will enable long range without extended advertising.

This has not happened since the release of the BT5.0 standard in December 2016, although the hardware of 90% of the BT controllers already allowed it. :P

pvvx, 2023-01-13:
I do not know for sure.  I can only guess...
The hardware of the chips does not care what synchronization to accept, since the modulation is the same for different PHY. The transmission frame header contains all the necessary information, and then the program's business.

pvvx, 2023-01-13:
On Linux, no patch is needed. You need to switch the adapter to encoded PHY and it will accept ads on LongRange. But others, standard BLE, will not be accepted.

DienoX, 2023-01-13:
If I run an encoded Bluetooth PHY in Linux and then connect to a thermometer with your software, will I get a greater range? Of course in your firmware I will choose LongRange BLE.

I found such a way to run an encoded PHY. Do you think he is good? I can't find anything better on Google on this topic, but I'm probably looking in the wrong place.
https://github.com/apache/mynewt-nimble/issues/893

pvvx, 2023-01-13:
> If I run an encoded Bluetooth PHY in Linux and then connect to a thermometer with your software, will I get a greater range? Of course in your firmware I will choose LongRange BLE.

Nothing will work - the program will switch the BT adapter to standard mode.

pvvx, 2023-01-13:
Wariant : primary BLE_PHY_1M, secondary BLE_PHY_CODED
```
		blc_ll_setExtAdvParam(ADV_HANDLE0,
				ADV_EVT_PROP_EXTENDED_CONNECTABLE_UNDIRECTED,
				adv_interval, adv_interval + 10,
				BLT_ENABLE_ADV_ALL, // primary advertising channel map
				OWN_ADDRESS_PUBLIC, // own address type
				BLE_ADDR_PUBLIC, // peer address type
				NULL, // * peer address
				ADV_FP_NONE, // advertising filter policy
				TX_POWER_0dBm, // TODO: advertising TX power cfg.rf_tx_power
				BLE_PHY_1M, // primary advertising channel PHY type
				0, // secondary advertising minimum skip number
				BLE_PHY_CODED, // primary advertising channel PHY type
				ADV_SID_0,
				0); // scan response notify enable 
		// debug!!!  Fix CodedPHY channel
		blc_ll_setAuxAdvChnIdxByCustomers(20); // auxiliary data channel, must be range of 0~36
```

[image]
(Random T & H for test)

---

Wariant : primary BLE_PHY_CODED, secondary BLE_PHY_CODED
Not accepted on standard adapters.
nRF Sniffer:
[image]

pvvx, 2023-01-13:
As a result, if LongRange is enabled, then it will be impossible to connect to this thermometer.
Switching format options on the fly does not allow the SDK and a lot of problems arise. Especially for ordinary users. It will raise a lot of questions and dissatisfaction :)

pvvx, 2023-01-13:
> I also agree that it would significantly impact power requirements. Because S=8 requires 8x the radio time for the same volume of data, because of the overhead of extended advertisement, and because it still needs to do legacy advertisement i parallel.

More than 8 times - advertising remains on the main channels +  data, on another channel.
[image]

hcgpalm, 2023-01-15:
> Nothing will work - the program will switch the BT adapter to standard mode.
Could you please clarify - what "program" are you talking about here?

> More than 8 times - advertising remains on the main channels + data, on another channel.
Yes, that's what I was trying to say.
My thought is that maybe there's some way to drastically reduce the coded PHY advertising interval?
Other long range/low power technologies like LoRaWAN, seems to use MUCH longer update intervals. Typically 5 minutes to an hour unless there's an event. Obviously for a good reason.

I agree that the device always needs to do 1M PHY advertising so that it's always visible and connectable for anyone. Switching advertising on primary channels to Coded PHY _only_ would essentially brick it for most users.

Since Telink seems to only support 1 advertising set, would it be possible to, for example, make it do legacy 1M PHY advertising just like today and then periodically switch to extended advertising with Coded PHY on primary channels?

I.e. using blc_ll_setExtAdvParam() to set up extended advertising using coded PHY on both primary and secondary channels and then calling blc_ll_setExtAdvEnable_n() to switch extended advertising on/off periodically.

pvvx, 2023-01-15:
> Since Telink seems to only support 1 advertising set, would it be possible to, for example, make it do legacy 1M PHY advertising just like today and then periodically switch to extended advertising with Coded PHY on primary channels?

No. Connecting the ext_adv module does not provide for rollback.
This is not possible without reversing binary libraries. Some of which I have already done...

In addition, switching ad types in ext_adv causes glitches. There are also glitches when using ext_adv type and a single ext_adv configuration. Sometimes does not include ads after disconnect, etc.

pvvx, 2023-01-15:
> Could you please clarify - what "program" are you talking about here?

About all implementations. Exception - DIY.

pvvx, 2023-01-15:
And there is a second problem with ext_adv:
ext_adv cannot be 'connectable' and 'scanable' at the same time. Requires type switching or having two ext_adv variants at the same time, which is not implemented in the SDK. (Switching can be done through a cold boot with different options for initializing drivers.)

pvvx, 2023-01-15:
Тестовая прошивка с поддержкой ‘Extended Advertising’ и ‘LongRange’ для LYWSD03MMC.
Файл https://github.com/pvvx/ATC_MiThermometer/blob/master/LR_v99.bin
На текущий момент эта  прошивка не поддерживает опции связанные с герконом.
После сброса питания прошивка работает в  стандартном режиме ‘1M PHY’.
Сброс и возобновление питания отключает опцию ‘Extended Advertising’ и ‘LongRange’.
Для включения режима ‘Extended Advertising’ в TelinkMiFlasher установите:
[image]
‘EXT.ADV’ без  ‘LongRange’ включает передачу рекламы в формате  LEGACY_CONNECTABLE_SCANNABLE_UNDIRECTED.
‘LongRange’ - включает передачу основной и дополнительной рекламы в формате ‘Coded PHY’ c S8 и варианте EXTENDED_CONNECTABLE_UNDIRECTED. 
Далее следует убедиться, что приемная сторона поддерживает ‘Extended Advertising’ в режиме ‘LongRange’ и если нет – вытащить и вставить батарейку.

DienoX, 2023-01-15:
Thank you very much for sharing the file.
Большое спасибо за то, что поделились файлом.

pvvx, 2023-01-15:
> Большое спасибо за то, что поделились файлом.

Исходники так-же обновлены с данными опциями.

pvvx, 2023-01-18:
> Any chance you could share your test code and procedure for flashing the Zigbee firmware onto a e.g. a CGG1?

https://github.com/pvvx/ATC_MiThermometer/issues/221#issuecomment-1279855367

jagheterfredrik, 2023-01-19:
I read the comment and I get that you don’t want to support Zigbee but being new to the platform it’d be great place to start.

Thank you for your hard work on the firmware!

jagheterfredrik, 2023-01-19:
E.g. The two-stage OTA you mention: do you mean we need to first flash a minimal zigbee firmware with only OTA support (<128k over BLE OTA) and then flash >128k main zigbee firmware over zigbee OTA?

pvvx, 2023-01-20:
С начала необходимо прошить малую прошивку BLE которая обрабатывает-очищает  Flash под OTA кода Zigbee. 
И только поле очистки производится загрузка кода ZigBee в совершенно другую область, чем BLE прошивки.
Zigbee код самой минимальной версии имеет более 128 килобайт и не может быть загружен стандартной BLE OTA от Telink.

DienoX, 2023-01-20:
Thanks @pvvx for the clarification, now I understand what you meant before.
Спасибо @pvvx за разъяснение, теперь я понимаю, что вы имели в виду раньше.

jagheterfredrik, 2023-01-20:
Thanks! So a BLE firmware that sets up flash according to dual mode and with a dual mode OTA procedure, then flash a <192k zigbee firmware?

pvvx, 2023-01-20:
http://wiki.telink-semi.cn/wiki/chip-series/TLSR825x-Series/
[ZigBee Development Manual](http://wiki.telink-semi.cn/doc/an/AN_19052900-E_Telink%20Zigbee%20SDK%20Developer%20Manual.pdf)
Figure 26: “512k Flash space allocation chart”
Bluetooth LE Development Manual ...
...

jagheterfredrik, 2023-01-21:
… thanks again for your firmware …

pvvx, 2023-02-09:
Для работы HA c LongRange желательно установить последний bluez и добавить в файле /lib/systemd/system/bluetooth.service
_ExecStart=/usr/local/libexec/bluetooth/bluetoothd_ **--experimental**

Перед запуском HA необходимо переключить USB-BT5.2 адаптер на CodedPHY:
_hcitool -i hci1 cmd 08 31 03 04 04_
Номер hci - свой, желательно от второго адаптера, а первый пусть принимает стандартные рекламы...
[image]

DienoX, 2023-02-09:
Thank you @pvvx for the useful post.
 I would like to understand one thing from him.  A Bluetooth 5.2 adapter is required for this to work, or is Bluetooth 5.0 enough?
 Asking because I thought CodedPHY was introduced with BT5.0.

Спасибо @pvvx за полезный пост.
 Я хотел бы понять от него одну вещь.  Для этого требуется адаптер Bluetooth 5.2 или достаточно Bluetooth 5.0?
 Спрашиваю, потому что я думал, что CodedPHY был представлен с BT5.0.

pvvx, 2023-02-09:
Достаточно BT5.0
[image]
_lsusb -s 5:2
Bus 005 Device 002: ID 0bda:8771 Realtek Semiconductor Corp. Bluetooth Radio_

pvvx, 2023-02-17:
> Спасибо @pvvx за полезный пост. Я хотел бы понять от него одну вещь. Для этого требуется адаптер Bluetooth 5.2 или достаточно Bluetooth 5.0? Спрашиваю, потому что я думал, что CodedPHY был представлен с BT5.0.

Да, с 2016 года так и продолжается "круговая порука" с BLE Long Range. Все ссылаются друг на друга.
Это типа "заговор" чтобы не дать народу использовать увеличение расстояния связи с BLE устройствами.
Примеры тут:
[So, I think the issue is with the Bluetooth integration then, isn't it? Perhaps @bdraco can help with this?](https://github.com/Bluetooth-Devices/bthome-ble/issues/45#issuecomment-1425562804)
[Home Assistant Bluetooth support is entirely based on bleak so we only support what bleak supports.](https://github.com/home-assistant/core/issues/88277#issuecomment-1433852281)
[Bleak not supports reading, writing and getting notifications from GATT servers, as well as a function for discovering BLE devices Bluetooth Core Specification Version 5.0 and next version?](https://github.com/hbldh/bleak/issues/1225)
И с моим знанием английского это бесполезно пробивать для пользователей.
Проще сделать версию только для русскоязычных...

pvvx, 2023-02-17:
В версии 4.2 будет включен полный "LE Long Range" для всех поддерживаемых термометров.
Обновление будет через пару часов, как отредактирую описания и [TelinkMiFlasher.html](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html).

closed 2023-02-18 by pvvx