pvvx/ATC_MiThermometer #224: Question: android command line app or with mqtt support?

olivluca, 2022-06-13:
Is there an android command line app that can be used to scan for the beacons? 
Alternatively is there an app that can read the temperature and humidity and sends it via mqtt?
I'm trying MijiaTemp but it only has support for thingspeak (besides, it's currently reporting 525.8ºC 😲).

pvvx, 2022-06-15:
?
[image]

olivluca, 2022-06-15:
[image]

olivluca, 2022-06-15:
I any case, even when it reads correctly, it doesn't support mqtt (though I'd prefer a command line utility and see the raw beacon)

pvvx, 2022-06-18:
In thermometers, user software uses 8 advertising formats. Switching is given in [TelinkMiFlasher.html](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html).

paolo-caroni, 2022-10-01:
There is https://github.com/G0RIAN/MijiaLogger but support only LYWSDCGQ

paolo-caroni, 2022-10-01:
Ok, there is also a fork that support mijia with ATC firmware https://github.com/bartashevich/MijiaLogger

closed 2023-03-28 by pvvx