pvvx/ATC_MiThermometer #225: Cannot apply configuration on FW 3.7

sybux, 2022-06-19:
Hi there,
I've just updated 1 Xiamo ATC with your new firmware (previously running ATC1441 firmware).
When I connect with your web interface to change config, after reading configuration, I change settings I want, and when I click on send config, I can see in log "Configuration send successfuly" but just after it says disconnected.
Just to be sure problem was not one the ATC, I've flashed a second one and same behavior.

I can provide any logs or other informations, just tell me which one and how to get them :)

Thx in advance

pvvx, 2022-06-20:
During tests, there is no shutdown when saving the configuration...
What settings are you changing?

sybux, 2022-06-21:
just trying to remove the emotion icon. After applying configuration, I'm disconnected of the ATC after 4-5 seconds

sybux, 2022-07-21:
any clue to try to understand why the device disconnect ?

Protoncek, 2022-07-21:
Did you try to change the battery? Use high quality one, not cheap china crap, since they tend to “crash” at bigger power. When communicating / sending config more power is needed and i guess that module restarts because of too low battery voltage…
I recently changed battery in one module, i used cheap one and brand new shows only just over 50%…

sybux, 2022-07-21:
Thx for reply. I've just tested with a fresh new ikea battery and it's working. previous battery was about 30%. So good point to configure use a new battery !

closed 2022-07-21 by sybux