pvvx/ATC_MiThermometer #226: Addition: two images of the packaging for the Xiaomi Mijia LYWSD03MMC info page

moritz-john, 2022-06-23:
Hello,

here are two images for the this site: https://pvvx.github.io/ATC_MiThermometer/ (only if needed)

I was unsure if I was buying the right version.
Turns out it was the right thermometer, so I wanted to add two more images of the packaging to make it easier for other people.
[image]
[image]

Kind regards.

pvvx, 2022-06-23:
The current firmware supports all the Xiaomi Mijia LYWSD03MMC released. There is no need to choose a production date and packaging options.

closed 2022-06-23 by pvvx

moritz-john, 2022-06-23:
> The current firmware supports all the Xiaomi Mijia LYWSD03MMC released. There is no need to choose a production date and packaging options.

To put it differently: I bought mine from here: https://www.amazon.de/gp/product/B08C7KVDJW/ and on the product page `LYWSD03MMC` is not mentioned even once. So at least for me a photo of the front packaging would have been helpful.

nagyrobi, 2022-06-23:
Same here. They are listed by the SKU number in the shops (NUN412...). Searching by term LYWSD03MMC doesn't bring up the product in webshops around here.

moritz-john, 2022-06-23:
Yes, they just listed as Xiaomi `Mi Temperature and Humidity Monitor 2`.