pvvx/ATC_MiThermometer #228: Не хватает мануала по восстановлению работы датчика в Mi Home после установки кастомной прошивки

IvanGerr, 2022-06-26:
Привет, 
В readme написано, что для восстановления работы датчика в MiHome после прошивки требуется: 
1. Подключить его в MiHome на стоковой прошивке
2. Прошить и восстановить binding key
3. Переименовать после прошивки в оригинальное название: LYWSD03MMC
4. Установить тип вещания MIJIA зашифрованное.

Исходя из того, что у меня не заработало, делаю выводы, что инструкцию не понял. 
Из моментов которые вызывают сомнение:  
1. "восстановить binding key" - куда восстановить? Он вроде бы остался на месте. По кнопочке "Show all mi keys" все те же данные, что в начале после кнопочки "do activation"
2. set advertising type to "MIJIA (MiHome)" marked "encrypted" - в выпадающем списке типов есть просто MIJIA,  а рядом галки AdFlags и Encrypted beacon. Обе чекнуты.
Прошу наставить на путь истинный, со своей стороны обязуюсь подготовить красочный мануал для таких же как я.

pvvx, 2022-06-30:
> 1. "восстановить binding key" - куда восстановить? Он вроде бы остался на месте. По кнопочке "Show all mi keys" все те же данные, что в начале после кнопочки "do activation"

Сначала термометр регистрируют в MI-Home.
Далее в программе [TelinkMiFlasher.html](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html) делают новую активацию. Это заменяет старый bindkey от Mi-Home на новый. И "do activation" необходимо сделать только один раз и записать ключи для повторного входа по "Login". (Если вам известен bindkey от Mi-Home, то возможно использовать "Login", без замены ключа.)
После записи custom-прошивки, по кнопочке "Show all mi keys", есть показ всех ключей. Предыдущий ключ там и будет отображен, как удаленный. Если "do activation" делали один раз, то кнопка "Swap previous token + bindkey" заменит ключи - текущий на предыдущий и обратно (предыдущий на текущий). Если много раз cделали "do activation", то в списке будет множество удаленных ключей - ищите ключ от Mi-Home.
Если вам известен bindkey от Mi-Home то просто впишите его в "EEP BindKey", а остальные удалите - "!Erase all Mi Keys!"
По п.п. 2 - всё ок.

Но предупреждаю: этими действиями вы нарушаете пользовательское соглашение, которое выводится при регистрации в Mi-Home. По этой причине работа custom прошивки в Mi-Home не популяризируется и публиковать "красочный мануал" я не буду.

closed 2022-09-12 by pvvx

TishSerg, 2023-08-03:
> 2\. Прошу наставить на путь истинный, со своей стороны обязуюсь подготовить красочный мануал для таких же как я.

Где ваш красочный мануал?

Ибо я вообще никакого мануала не видел, кроме этого треда здесь.