pvvx/ATC_MiThermometer #230: Cannot graph any data from CDGK2 with FW 3.7

leematos, 2022-07-03:
Love the firmware so far.

```
9:50:23 AM: Hardware Version: CGDK2 2.1.0, Software Version: 3.7, Sensor: SHTC3
9:50:23 AM: Custom config: [19, 96, 0, 0, 40, 4, 169, 124, 49, 134, 60]
```

I can connect via Telelink flasher, and if I read the last 50 records there, I can see output.

I cannot connect to any ATC graphs or graph memo. I'm not sure what I'm missing.

I've tried by connecting to telelink first (bluetooth symbol on device shows connected) and I've also tried connecting "directly" from the graph pages. What am I doing wrong?

Happy to help document it so it's clearer once we figure it out :D

leematos, 2022-07-03:
Also using recent chrome and I'm on a Mac if that matters.

pvvx, 2022-07-03:
This is due to the specifics of Apple. I don't use Apple...

zpon, 2022-07-03:
I'm seeing the same with Chrome on Android

leematos, 2022-07-04:
@pvvx Thanks -- This diff fixed it, happy to make a PR for it if you'll accept. I copied the device options from Telelink flasher, and once I did that I was able to finally see my 3 CDGK2 sensors (along with many other entries for "unsupported devices". I don't know enough about Bluetooth settings to hide those). 

```
diff --git a/ATC_MiThermometer/GraphMemo.html b/ATC_MiThermometer/GraphMemo.html
index 9f54f3d..4fd14c1 100644
--- a/ATC_MiThermometer/GraphMemo.html
+++ b/ATC_MiThermometer/GraphMemo.html
@@ -368,9 +368,9 @@ function connect() {
 }
 // Запрос выбора Bluetooth устройства
 function requestBluetoothDevice() {
-       var deviceOptions = {optionalServices: [0xfe95, 0x181a, 0x1f10]};
-       deviceOptions.filters = "ABCDEFGHIJKLMNOPQRSTUVWXYZ abcdefghijklmnopqrstuvwxyz_#@!*';,.<>{}[]"
-       .split("").map((x) => ({namePrefix:x}));
+                var deviceOptions = {
+                optionalServices: ['00010203-0405-0607-0809-0a0b0c0d1912', 'ebe0ccb0-7a0a-4b0c-8a1a-6ff2997da3a6', '22210000-554a-4546-5542-46534450464d', 0x180a, 0x181c, 0x181e
, 0xfe95, 0x1f10, 0x1f11, 0x181a, 0xfff9, 0xfdcd, 0xffe0],
+                acceptAllDevices: true };
```

leematos, 2022-07-04:
I could probably add this filter to hide the unsupported devices: https://github.com/pvvx/pvvx.github.io/blob/master/ATC_MiThermometer/TelinkMiFlasher.html#L125-L128

closed 2023-02-18 by pvvx