pvvx/ATC_MiThermometer #231: Xiaomi Mijia (LYWSD03MMC) disconnecting

Tihomir971, 2022-07-05:
My devices (4) disconect after connecting. I cant configure or update firmware.
13:30:35: Hardware Revision String: B1.4
13:30:35: Detected custom Firmware
13:30:35: Hardware Version: Unknown B1.4, Software Version: 3.4, Sensor: SHTC3
13:30:35: Custom config: [37, 0, -2, 0, 40, 4, 169, 124, 49, 184, 60]
13:30:45: Load firmware file '?'...
13:30:45: File: undefined
13:30:45: Disconnected.
13:30:51: Reconnect
13:30:54: Hardware Revision String: B1.4
13:30:54: Detected custom Firmware
13:31:04: Disconnected.

pvvx, 2022-07-09:
[correct hw if ver < 3.6](https://github.com/pvvx/pvvx.github.io/commit/83b2120142b482002b58a62a5950353803b43fdb)

Tihomir971, 2022-07-09:
> [correct hw if ver < 3.6](https://github.com/pvvx/pvvx.github.io/commit/83b2120142b482002b58a62a5950353803b43fdb)

It's same with other:
10:38:31: Connecting to: OfficeF251
10:38:37: Hardware Revision String: B1.4
10:38:37: Detected custom Firmware
10:38:42: Hardware Version: LYWSD03MMC B1.4, Software Version: 3.7, Sensor: SHTC3
10:38:42: Custom config: [37, 0, 0, 0, 160, 6, 169, 124, 49, 128, 60]
10:38:52: Disconnected.

pvvx, 2022-07-11:
The battery does not have enough charge to connect. Try replacing with a new battery.

Protoncek, 2022-07-12:
The battery can be new, but “not good”. Been there, done that. For this device only good quality batteries are “good enough”, because spike current draw can be bigger. Definitely not cheap chinese ones. 
For example: rf remotes (for automatic gates): they ussually don’t work with new chinese batteries - not even right after batteries are inserted. When button is pressed voltage just drops to nearly zero.

Tihomir971, 2022-07-12:
problem solved

fanoush, 2022-10-04:
> Been there, done that. For this device only good quality batteries are “good enough”, because spike current draw can be bigger. Definitely not cheap chinese ones.

If you open it ,there are empty pcb pads where capacitors normally should be. Check https://www.ti.com/lit/wp/swra349/swra349.pdf
Adding capacitors could improve battery life and solve this issue too.

closed 2023-02-18 by pvvx