pvvx/ATC_MiThermometer #233: Customer firmware no longer connected to Xiaomi gateway 3

stelectronic, 2022-07-08:
I register the Xiaomi Mijia LYWSD03MMC in MiHome app and connected to Xiaomi gateway first. Then flash to custom firmware and restore the Mi Token &  Mi Bind Key as original firmware.  Then replacing the device name with "LYWSD03MMC" and setting the advertising type to "MIJIA (MiHome)" marked "encrypted".
[image]
[image]
Everything is OK except Mihome app and Xiaomi gateway cannot receive its message, so I cannot do some auto-flow per the temperature....

stelectronic, 2022-07-08:
[image]
Should this be marked? Appreciated for your guidance...

pvvx, 2022-07-09:
> Everything is OK except Mihome app and Xiaomi gateway cannot receive its message, so I cannot do some auto-flow per the temperature....

It is necessary to restore the correct bindkey from MiHome.

Receiving data in HomeAssistant from BT adapter:
[image]
Getting data in HomeAssistant from Gateway 3:
[image]
[image]

MiHome will not connect to the thermometer. Works only through the gateway.

> Should this be marked? Appreciated for your guidance...

https://github.com/pvvx/ATC_MiThermometer/issues/221

stelectronic, 2022-07-10:
@pvvx  I noticed that you fill the blank of EEP Bindkey, should I use the value of Mi Band Key of original firmware here and then click Set EEP Bindkey? Thanks!
[image]

pvvx, 2022-07-11:
The bindkey written in Flash in Mi format takes precedence.
If the Mi keys in Flash are removed, then the bindkey from the EEP is used.

WarC0zes, 2022-09-15:
Hi, 
I have trouble understanding the Bindkey part. 
I clicked on get EEP Bindkey and there was no key. I copy the Bindkey from above, paste and Set EEP Bindkey. But the thermometer does not send the information to gateway 3.
I checked Mijia (mi home) and encrypted beacon. Rename the thermometer to LYWSD03MMC.

What am I doing wrong?

[image]

WarC0zes, 2022-09-15:
@stelectronic You have managed to make it work with the xiaomi gateway 3?

pvvx, 2022-09-15:
**TelinkMiFlasher**
[image]

**Home Assistant: Passive BLE Monitor integration**
[image]

**Home Assistant: Xiaomi Gateway 3**
[image]

[image]

[image]

**nRF Connect**
[image]

pvvx, 2022-09-15:
>  I copy the Bindkey from above, paste and Set EEP Bindkey. 

Bindkey must be registered in MiHome on a thermometer with official firmware.
Only after that the device is flashed and the already registered bindkey is entered into it.

WarC0zes, 2022-09-16:
@pvvx Thank you for your answer, I understand the procedure better. Thank you for your awesome work.

WarC0zes, 2022-09-16:
> If you already have a MiHome bindkey, then use "login" by entering the MiHome bindkey.
I did it this way, it works. You are a boss, thanks.
[image]

closed 2023-02-18 by pvvx

Presjar, 2024-01-27:
Hello.

Could I please get an instruction on how to correctly complete the identification of the original Mi Token and Mi Bind key?

This is what I have done so far in attempting to get FW1.6 flashed LYWSD03MMC to send data to the Gateway 3. Have I missed any steps? I don't seem to be getting any updates via the Gateway showing in the app after flashing. Thank you.

1. Restored Original FW from https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html
2. Added LYWSD03MMC to Xiaomi Home Android App. Note I have a Gateway 3 working as Gateway.
3. Go back to https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html
4. Connect to  MAC address of LYWSD03MMC
5. Select Do Activation
6. Select to flash firmware 1.6. Wait for complete
7. Connect to flashed LYWSD03MMC 
8. Select show all mi keys
9. Logs show 
`Marked as delete Key0: miToken: A2DB9C3<REDACTED>317668D, miBindKey: B9B2D6930C9C<REDACTED>5327B4EACDFDF`
10. Enter the log shown miToken into the web ui for mi Token
11. Enter the log shown miBindKey into the web ui for mi Bind Key
12. Click Set new Token and Bind Key
13. Enter log shown miBindKey into Web ui EPP BindKey
14. Click Set EPP BindKey
15. Use Web ui to restore the device name to LYWSD03MMC
16. Use Web ui config area to select Advertising Type to MIJIA and Check Encrypted Beacon
17. Click Send Config
18. Disconnect and Reconnect
19. Go to keys area and click show all mi keys
20. The keys shown in the UI and the values saved previously.

I see in above screenshot  the BT5 and CSA2 mode is selected, do I need this?

pvvx, 2024-01-27:
> I see in above screenshot the BT5 and CSA2 mode is selected, do I need this? I can see different selection on my web ui "BT5+ PHY and LE Long Range (only for BT5.0+ adapters!)

Do not install.

> Do I also need to uncheck AdFlags ?

Select checked.

Presjar, 2024-01-28:
Thank you for the confirmation.

I was able to get two LYWSD03MMC working with Gateway 3 following the steps posted above. I Ensured AdFlags was selected and un-ticked the BT5 settings.

I setup my third LYWSD03MMC later, and followed your suggestion of setting EPP bindkey only, however Gateway 3 would not show updated sensor readings. When I went back to the settings and completed steps 10, 11 and 12, Gateway 3 now gets sensor updates.

Thanks for your help.