pvvx/ATC_MiThermometer #234: [Feature request] Add mode without BLE: turn off after 30 min

popolite, 2022-07-12:
Count 1800 seconds after powered on and stop BLE to reduce battery consumption and protect against connection.

pvvx, 2022-07-13:
> Count 1800 seconds after powered on and stop BLE to reduce battery consumption and protect against connection. И еще можно сделать включение на пол часа после повышения влажности больше 90% за несколько секунд (подышать). 

Не имеет большого смысла. Смотри диаграмму потребления в зависимости от adv. интервала:
[image]

Ток потребления устройством в режиме сна (Xiaomi LYWSD03MMC, HW:1.4) около 9+ мкА. 
При установках по умолчанию - до 14 мкА.

>Достаточно ли будет для отключения: bls_ll_setAdvEnable(0)?

Нет. При отключении AdvEnable SoC больше не выйдет из режима сна. Будет отключен power-management.
Придется вызывать принудительный deep-sleep c ограниченным в SDK интервалом.
Это дает много последствий - сильно большой и не рассчитываемый уход времени и изменение многих основных алгоритмов работы программы. 
Старт из полного deep-sleep приводит к потере всей памяти и длительному старту циклу инициализации (большому потреблению) и имеет смысл вызов только на большой временной интервал, а он ограничен в SDK.

pvvx, 2022-07-13:
> И еще можно сделать включение на пол часа после повышения влажности больше 90% за несколько секунд (подышать).

После нескольких повышений влажности более 90%, через пару месяцев окислятся контакты батарейки и термометр отключится или будет глючить – пропадать передачи от него. Так-же окислятся контакты платы с резиновой токопроводящей прокладкой к LCD экрану. И контакты на самом LCD аналогично. При этом, если тока через контакты не будет или он будет мал, тогда такие эффекты (окисление контактов) появляются быстрее.
Проверено на нескольких LYWSD03MMC устанавливаемых в парнике.

xyyx, 2023-01-19:
> > И еще можно сделать включение на пол часа после повышения влажности больше 90% за несколько секунд (подышать).
> 
> После нескольких повышений влажности более 90%, через пару месяцев окислятся контакты батарейки и термометр отключится или будет глючить – пропадать передачи от него. Так-же окислятся контакты платы с резиновой токопроводящей прокладкой к LCD экрану. И контакты на самом LCD аналогично. При этом, если тока через контакты не будет или он будет мал, тогда такие эффекты (окисление контактов) появляются быстрее. Происходит не замыкание, а окисление и для пробоя окисла требуется ток... Проверено на нескольких LYWSD03MMC устанавливаемых в парнике.

Доброго времени суток. Мне кажется, я с этим столкнулся. Стоит в подвале гаража. Там 90% часто. В качестве шлюза esp32. Просто перестал нормально слать данные. При этом на экране все есть. Сменил батарейку, не помогло. Что можно сделать? Промыть спиртом? При этом через флешер нормально обновился(

closed 2023-02-18 by pvvx