pvvx/ATC_MiThermometer #236: Cannot activate after flashing original firmware via uart

rocket59, 2022-07-18:
As I managed to brick 2 of my thermometers originally I successfully reflashed the original 109 firmware (as they were 1.4 boards).

However when I open the telink flasher webpage and connect to the thermometers it says custom firmware detected and refused to go any further.

Any ideas?

rocket59, 2022-07-18:
Update: I managed to connect with one of the thermometers but when I try to activate it I get an error message saying "Update error:some error while sending char data".

pvvx, 2023-02-18:
You are confused in different versions of firmware and utilities. There is nothing to help here.

closed 2023-02-18 by pvvx

rocket59, 2023-02-18:
Don't think so, pretty sure it was done correctly. Anyway this was a while ago, in the end these thermometers were an interesting experiment but I'm back to using zigbee sensors.