pvvx/ATC_MiThermometer #237: Request: Disable Screen

AleXSR700, 2022-07-29:
Hello,
I am using multiple Xiaomi Mijia Thermometer 2 BLE and all with your custom firmware.

Since I am using them almost exclusively as sensors connected to Home Assistant, I wondered if you could add an option to disable the display?
That would save some battery :)

Thank you
Alex

bentolor, 2022-08-14:
Have you seen & read https://github.com/pvvx/ATC_MiThermometer#average-power-consumption @AleXSR700 ?

The actual power consumption is neglectible in comparison to the self-discharge of the CR2033 cells. Most power will be used on RX/TX anyways.

mat-c, 2022-08-28:
What about devices used only for logging ?
What is the consumption of lcd ? Does it depends of number of segment displayed ?
What about the impact of refresh rate ?

What do you mean by  self-discharge of the CR2033 cells ? Some battery have low self-discharge rate and be good after several year. For example https://data.energizer.com/pdfs/cr2032.pdf is rated 1% self-discharge per year.

pvvx, 2022-09-13:
There are 3 types of LCD driver ICs used in different editions of LYWSD03MMC.
Documentation for the used models of LCD driver chips is not in the public domain.
For other thermometer variants that use E-Ink driver ICs, there is only one applicable document. When programmatically switching the driver chip to sleep mode, the consumption is still more than that of the main SoC.
Reducing the consumption of the LCD driver chip is possible only in the event of a physical power off or desoldering the chip.

mat-c, 2022-09-13:
Thanks for your feedback.
On other issue https://datasheet4u.com/datasheet-parts/BU9792FUV-datasheet.php?id=705112 lcd datasheet is show. This lcd driver is not used on all the LYWSD03MMC edition ?

Somebody implemented something on https://github.com/unaiur/ATC_MiThermometer/blob/master/src/display_drv_cgdk2.c#L109

It use DISPLAY ON/OFF bit in MODE SET register. This will work only on specific devices ?

pvvx, 2022-09-14:
> Thanks for your feedback. On other issue https://datasheet4u.com/datasheet-parts/BU9792FUV-datasheet.php?id=705112 lcd datasheet is show. This lcd driver is not used on all the LYWSD03MMC edition ?

This controller is used only in CGDK2:
https://pvvx.github.io/CGDK2/
https://pvvx.github.io/CGDK2/CGDK2-2/

> It use DISPLAY ON/OFF bit in MODE SET register. This will work only on specific devices ?

The "power save mode3" command does not provide what you want. The consumption of this controller is still more than that of the sleeping TLSR825x. And for hardware shutdown between chips in CKDK2 there are no necessary connections.

In all other models of thermometers, the communication protocols with the display driver have been looked from the official version by a digital analyzer-logger. And then partially modified to reduced consumption ...

AleXSR700, 2022-09-16:
Thank you @pvvx for all the technical and background information.

So is there one true set of optimal settings?

I would really love to reduce the frequency of BLE publish (even if it saves little power) but from my understanding this is not possible, correct? It will become unstable when I try to go for 30 or even 60 second intervalls.

pvvx, 2022-09-16:
The maximum interval according to the specification is 10 seconds. On all modern smartphones with Android, this works fine. On linux - trouble and a lot of torment, on Windows - it depends on the BT adapter.
And increasing the interval more than 10 seconds does not give an adequate increase in battery life. This is shown by the simplest calculation presented in the graphs:
Average power consumption -> https://github.com/pvvx/ATC_MiThermometer#average-power-consumption

AleXSR700, 2022-09-17:
Well, an additional ~10 months at 60 sec would sound significant to me :)

pvvx, 2022-09-17:
The source code is given - collect your version :)

AleXSR700, 2023-01-21:
I was taking a look at my Mijias and extending the battery life will would be neat.

[image]

Unfortuantely I do not know how to increase the transmission time to increase the battery life by the theoretical 10 months.
Could someone help me, please?

closed 2023-02-18 by pvvx