pvvx/ATC_MiThermometer #239: Issue with pvvx custom payload

Dylan96, 2022-08-10:
I have a standard LYWSD03MMC v1.4 with the latest 3.7d custom firmware and i set the custom pvvx advertising format. The problem is i can't reliably read humidity and battery level.
This is one random payload i got:  
`0x10161A18A4C138690CFF012728570BA8E20B094154435F363930434646`

```
size: 10

uid: 16

UUID: 1A18

MAC Address: A4C138690CFF

Temperature: 0127 (29.5C)

Humidity: 28 (40%, no decimal?)

Voltage: 570B (2903 Volts)

Battery Level:  ??

Counter:  ??

Device Name: 4154435F363930434646 (ATC_690CFF)
```
Where are the humidity decimals? am i missing something? thank you.

pvvx, 2022-08-13:
This is the atc1441 format.

Dylan96, 2022-08-13:
Wow you are right. I guess my config didn't save correctly twice...

closed 2022-08-13 by Dylan96