pvvx/ATC_MiThermometer #240: Memo: "Delete all records" does not delete ALL records

bentolor, 2022-08-14:
Hi @pvvx Victor,

I'm using the Graph Memo functionality quite regularily. Unfortunately the delete all record function _does not delete all records_. 

See below after a fresh "delete all records". It only seems to delete all recent records.

## Example 1: After "delete all records"
[image]

## Example 2: Before "delete all records"
[image]

## Example 2: After "delete all records"
[image]

pvvx, 2022-08-17:
It is possible that the battery is low.
There was not enough energy for the entire memory erasing cycle and the device rebooted?

bentolor, 2022-08-18:
@pvvx I don't think so as I'm able to reproduce this behaviour on multiple devices and I'm not loosing bluetooth connection. I can directly issue a "get time" command right after it. 

Just did it on a 3rd device: It always seems to be based on very old data. What remained are the values from 30 March to 22 April. Deleted are all newer than April 22 18:00.

Also: Repeated deletion does not change the reminder area: It stays the same date range: 30 March to 22 April.

I'd see some display flickering or loss of bluetooth connection in case of a reboot, right?

pvvx, 2022-08-19:
> I'd see some display flickering or loss of bluetooth connection in case of a reboot, right?

Yes.
The procedure code in the loop erases the sectors.
It cannot abort until it has erased all Flash sectors associated with "Memo".

bentolor, 2022-08-25:
@pvvx Just tried this again on _another_ device and now **I feel pretty confident that its not the battery**. Here is why:

* Downloading the full data takes much, much, much longer and I can even see the display become lighter during the transmission while
* deletion happens in a glimpse: Display is stable; no flicker; bluetooth stays nothing at all.

Please check the attachments: First on before deletion, second after deletion. Maybe they can give a hint.
[lounge-measurements.csv](https://github.com/pvvx/ATC_MiThermometer/files/9424410/lounge-measurements.csv)
[lounge-measurements-truncated.csv](https://github.com/pvvx/ATC_MiThermometer/files/9424412/lounge-measurements-truncated.csv)

Looking at the referred code and asking a few dumb questions:

* Why [start at`FLASH_ADDR_START_MEMO + FLASH_SECTOR_SIZE` vs. `FLASH_ADDR_START_MEMO`?](https://github.com/pvvx/ATC_MiThermometer/blob/master/src/logger.c#L119)
*  [Was `MEMO_SEC_ID` already present in previous firmwares?](https://github.com/pvvx/ATC_MiThermometer/blob/master/src/logger.c#L123). I might have had an older firmware on collecting that old data
* And I don't get it: Why only erase if *no* known sector is present? You told mi it would erase _all_ sectors independent from any condition. This does not seem to be true.
```
		if (tmp != MEMO_SEC_ID)
			_flash_erase_sector(faddr);
```

closed 2022-09-12 by pvvx

bentolor, 2022-09-12:
Thank you, @pvvx! I just tested the new 3.8 on one device and can confirm that deleting now works as expected! Awesome!