pvvx/ATC_MiThermometer #241: Can't flash firmware on B1.4 version

TaiJiGong, 2022-08-18:
Can't flash firmware on B1.4 version. 

Detected HW version is 0000. 

[image]
[image]

TaiJiGong, 2022-08-18:
Status: LYWSD03MMC HW:0000 Connected, you can now Do the Activation to either get the Token or flash a new Firmware

TaiJiGong, 2022-08-18:
11:19:03: Searching for devices
11:19:10: Connecting to: LYWSD03MMC
11:19:13: Hardware Revision String: 0000
11:19:13: Detected Mi device
11:19:13: Connected
11:19:49: Activating now, please wait...
11:19:53: Registration successfull
11:19:53: Send Login, please wait...
11:19:53: Received device infos are correct
11:19:54: Login successfull

TaiJiGong, 2022-08-18:
There is no selection for the firmware. 
only a selection for a file.

TaiJiGong, 2022-08-18:
I'm downloading the firmware bin and then uploaded to the device. 

That's working. 

12:06:50: Searching for devices
12:06:54: Connecting to: LYWSD03MMC
12:06:59: Hardware Revision String: 0000
12:06:59: Detected Mi device
12:06:59: Connected
12:07:06: Activating now, please wait...
12:07:10: Registration successfull
12:07:10: Send Login, please wait...
12:07:10: Received device infos are correct
12:07:10: Login successfull
12:08:01: File: ATC_v37d.bin
12:08:01: File size: 67252 bytes
12:08:01: Count: 4204
12:08:03: Start DFU
12:08:55: Update done after 51.5 seconds
12:09:01: Disconnected.
12:09:45: Reconnect
12:09:53: Hardware Revision String: B1.4
12:09:53: Detected custom Firmware
12:09:56: Hardware Version: LYWSD03MMC B1.4, Software Version: 3.7, Sensor: SHTC3
12:09:56: Custom config: [5, 96, 0, 0, 40, 4, 169, 124, 49, 128, 60]
12:10:15: Send new DevTime ok
12:10:16: Device Date: 2022-08-18 12:10:15.000
12:10:21: New custom config: [5, 96, 0, 0, 40, 4, 169, 124, 49, 128, 60]
12:10:21: Settings 55056000002804A97C31803C was send successful
12:10:21: Hardware Version: LYWSD03MMC B1.4, Software Version: 3.7, Sensor: SHTC3
12:10:21: Custom config: [5, 96, 0, 0, 40, 4, 169, 124, 49, 128, 60]

pvvx, 2022-08-19:
> There is no selection for the firmware. only a selection for a file.

This means that this firmware version from the manufacturer does not support identification.

gnbl, 2022-08-21:
Hi, I had the same issue and uploaded the custom release firmware manually - after that, detection works.
It would be nice to either show a hint in the Flasher, and/or add a note to the readme explaining this issue (https://github.com/pvvx/ATC_MiThermometer/blob/master/README.md?plain=1#L69 and https://github.com/pvvx/ATC_MiThermometer/blob/master/README.md?plain=1#L109).

T3rm1, 2022-09-09:
After I click "Do activation" the log shows "Activating now, please wait..." and the display of the device changes. After 4 seconds it says "Disconnected." in the log. Any ideas?

closed 2023-02-18 by pvvx