pvvx/ATC_MiThermometer #242: Telink Flasher (https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html) won't show or allow setting confirm after flash

akosiaris, 2022-08-26:
Hi,

Let me start by saying thanks for this set of tools. 

I am meeting the following problem. After obtaining a couple of `LYWSD03MMC`, I straight through the [Telink Flasher](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html). Device was detected and per the suggestion I applied firmware `3.7` fine. After removing and adding the battery in the device and trying to connect again to apply configuration (which is my understanding I should be able to do via this tool), I meet the following screen:

[image]

I don't seem to be able to do much in that screen. I am guessing that's not right.

Platform: Linux, Chrome `104.0.5112.79` with `#enable-experimental-web-platform-features` enabled.

Protoncek, 2022-08-26:
I think that you flashed atc firmware, but now you're trying to connect with pvvx flasher (it says atc firmware found). 
Either connect with atc flasher or upload pvvx firmware.

akosiaris, 2022-08-26:
> I think that you flashed atc firmware, but now you're trying to connect with pvvx flasher (it says atc firmware found). Either connect with atc flasher or upload pvvx firmware.

I 'd wish. That would explain it, but I stayed squarely in the pvvx flasher and nothing else. Straight from the box, to pvvx flasher, do activation, flash the ATC_37d.bin firmware the pvvx flasher proposed.

TimeTravel-0, 2022-08-26:
I can confirm this behavior. I was able to flash a brand new device via pvvx web flasher (just used the suggested firmware), but no settings show up afterwards, just "Setting 55 was sent successful".

I use Chromium Version 104.0.5112.79 (Official Build) built on Debian 11.4, running on Debian 11.4 (64-bit) with the experimental features enabled.

nRF connect DIS output:
[image]

Browser screenshot, including console:
[image]

Protoncek, 2022-08-26:
Interesting... Did you guys try atc's flasher page? Just "in case"... since it points you to TelinkOTA.html... i'm interested to know if there settings will be available (it would point to "forgotten" firmware name change).

TimeTravel-0, 2022-08-26:
I just flashed https://github.com/pvvx/ATC_MiThermometer/blob/master/Original_OTA_Xiaomi_LYWSD03MMC_v1.0.0_0130.bin using https://pvvx.github.io/ATC_MiThermometer/TelinkOTA.html to get back to stock firmware. This worked.

Resulting DIS: values:
[image]

the atc flasher reported:
```
Log:
17:22:57: Searching for devices
17:23:23: Connecting to: LYWSD03MMC
17:23:30: NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
17:23:30: Reconnect 1 from 5
17:23:30: NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
17:23:30: Reconnect 2 from 5
17:23:30: NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
17:23:30: Reconnect 3 from 5
17:23:30: NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
17:23:30: Reconnect 4 from 5
17:23:30: NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
17:23:30: Reconnect 5 from 5
17:23:30: NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
17:23:30: Something went wrong, to many reconnect's
```
I experienced bluez not being happy about changed services on the same mac address in the past, "remove <mac>" via bluetoothctrl did not help here. It is possible that the same thing happens after flashing pvvx firmware. I'll play with this a bit and see if I can make bluez find the right services/characteristics again.

akosiaris, 2022-08-26:
Tried from a chrome on an Android. Everything worked out fine. This seem specific to chrome on Linux, so I am updating the issue subject. 

Couple of things I tried. Stopping starting bluetooth.service with systemctl, cancel-pairing command of bluetoothctl. No diff

Stolz, 2022-11-04:
I have the exact same problem. Chorme + Gentoo Linux. After successfully flashing the suggested firmware (ATC_v38.bin), the device name changes from  LYWSD03MMC to  ATC_362FFF, and TelinkMiFlasher will be able to connect but no config options are shown

sezlony, 2022-12-03:
just ran into this problem using android phone and I don't really understand what is the solution to avoid bricking the device. :(

please, could someone tell me what I should do now? I don't want to worsen the situation...

oiesauvage, 2022-12-14:
@akosiaris Your issue seems similar to #265. Setting  `devIdEnabled = true;` arround line 1320 of the webapp `pvvx.github.io-master/ATC_MiThermometer/TelinkOTA.html` (in local) seemed to solve the issue on my side (chromium on linux).

pvvx, 2023-02-18:
These are problems of Linux - all old versions of kernels and Bluez.
Linux still does not have support for Bluetooth 5.0, released in 2016. Although BT adapters can work with Bluetooth 5.0 from 2016.

closed 2023-02-18 by pvvx