pvvx/ATC_MiThermometer #243: [feature request] The battery sensor attribute.

sp7dpt, 2022-08-28:
The battery sensor has a battery health percentage, but has no battery attribute in the list. Can you add it?
[image]

Protoncek, 2022-08-28:
What do you mean? It clearl states “device_class” as battery and “unit of measurement” as %…

sp7dpt, 2022-08-28:
Other sensors in Home assistant have a battery attribute.
For example, the Zigbee motion sensor

[image]

I have automation that polls all battery-powered sensors and sends notifications about low battery.

Protoncek, 2022-08-28:
I don’t see any practical reason why you’d have two exactly same values in one sensor. In your example you also have 77.5 value as “state”. Why again same state as attribute? (Just wondering…)

sp7dpt, 2022-08-28:
It seems to be a standard. See the integration that shows the battery status https://github.com/maxwroc/battery-state-card
It is based on the battery attribute and my my automation too.

Protoncek, 2022-08-28:
Hm...
i installed above addon for test and my 8 xiaomi's thermometer batteries show perfectly in this card even without "battery" attribute. I think that "attribute" setting in this card is just an option, if battery state is not defined in "main state". 

[image]
[image]

sp7dpt, 2022-08-28:
OK, maybe that's a bad example. I just wanted to show you that the sensors have duplicate values. I do not know why.
Another example of an Aqara sensor
[image]

closed 2023-02-18 by pvvx