pvvx/ATC_MiThermometer #244: Name change of HA BLE format to BTHome

Ernst79, 2022-09-01:
HA BLE format is one of the advertising formats that is available in the pvvx firmware. In Home Assistant 2022.9 we have added support for HA BLE as a core integration, based on the parser in BLE monitor. During the development, we (the HA devs and me) decided that we prefer to have a new name for HA BLE, namely **BTHome**. We also added a website with all the info (https://bthome.io)

BLE monitor 10.0.0 and the new core integration that will become available have been updated to use the new name BTHome. 

Is it possible to update the references/names from HA BLE in the firmware and documentation?

pvvx, 2022-09-01:
ok

Ernst79, 2022-09-01:
The HA integration won’t have the binary sensors yet, these will follow in October release. BLE monitor still have the binary sensors of course.

closed 2022-09-08 by pvvx