pvvx/ATC_MiThermometer #246: Does the device loose its temperature calibration when flashing to the Custom firmware?

Allan-Hoejberg, 2022-09-10:
I have 8 MiThermometer 2´s, and a reference termometer.
Before flashing to the custom firmware the MI device show about the same temperature as my reference termometer.
But after I flash the customer firmware on, the sensores typically show +0,3-0,6 more the my reference termometer.
and also the custom flashed devices, seem to be very jumpy, showing 24.1 and after a bit showing 23.8, and then maybe back to 24,1.

Timminater, 2022-10-25:
Same problem here, fluctuation of almost a whole degree C when using the BThome format

rubencm, 2022-11-12:
I am not sure if is the same problem, but after upgrading to 3.8 and change to BTHome, the sensor is reporting extreme temperatures. I dont have that problem with humidity.

[image]

DrSchottky, 2022-11-14:
> I am not sure if is the same problem, but after upgrading to 3.8 and change to BTHome, the sensor is reporting extreme temperatures. I dont have that problem with humidity.
> 
> [image]

Having the same problem, I'm getting a wrong temp read from HA once a day

pvvx, 2022-11-19:
https://github.com/1technophile/OpenMQTTGateway/issues/1302#issuecomment-1302736929

rubencm, 2022-11-19:
[image]

This only happened after moving to version 3.8 + BTHome, i am using now chinese batteries, but i dont know if that have anything to do with it

pvvx, 2022-11-19:
ESPs have always been distinguished by this - there is not enough memory for the full-fledged operation of the TCP, plus the use of C ++ without MMU (Memory Management Unit) for heap memory = inevitable falls, crashes, errors ...

pvvx, 2022-11-19:
> I have 8 MiThermometer 2´s, and a reference termometer. Before flashing to the custom firmware the MI device show about the same temperature as my reference termometer. But after I flash the [...]

Mi firmware measures and transmits once every tens of minutes. The user firmware measures and transmits values every 10 seconds.
Reconciliation with another thermometer requires a special oven and waiting hours for the temperature probes to balance.
The inherent noise of the applied temperature sensors, together with all the specifics of the circuitry, is at the level of + -0.02С.

Thermometer test in a box. It still depends on how close the person is...
[image]

DrSchottky, 2022-11-19:
> ESPs have always been distinguished by this - there is not enough memory for the full-fledged operation of the TCP, plus the use of C ++ without MMU (Memory Management Unit) for heap memory = inevitable falls, crashes, errors ...

In my case there're no ESPs involved: Sensor HW B1.4, SW:3.8, format BTHome. Data read from a RPi 3B running latest HA w/ BTHome integration

rubencm, 2022-11-20:
In my case neither, I have the devices connected directly to home assistant with BTHome

pvvx, 2022-11-20:
Home Assistant, BTHome integration, connected to an optional USB-BT5.0 adapter:
[image]

DrSchottky, 2022-11-20:
Checked some logs and this looks more like a sw problem related to Pi's BT rather than an issue with this fw. @rubencm are you using Pi's embedded BT radio too?

rubencm, 2022-11-21:
> Checked some logs and this looks more like a sw problem related to Pi's BT rather than an issue with this fw. @rubencm are you using Pi's embedded BT radio too?

Yes

pvvx, 2022-11-21:
The software limit of readings from the sensor is -129.99С to +45.00С even if its data is error. Limited by the register value from 0 to 0xFFFF. A failure (0xFFFF) is checked and discarded. The checksum of the data transmission from the sensor is also checked.
https://github.com/pvvx/ATC_MiThermometer/blob/master/src/sensors.c#L141

rubencm, 2022-11-22:
Then is not an issue of the sensor? because [here](https://pastebin.com/r1tPEbYV) for example, in home-assistant, i have a reading of -316.5 (outside of its limits), also i have seen some times where is just "unavailable".

DrSchottky, 2022-11-22:
> Then is not an issue of the sensor? because [here](https://pastebin.com/r1tPEbYV) for example, in home-assistant, i have a reading of -316.5 (outside of its limits), also i have seen some times where is just "unavailable".

According to my tests there are errors at UART level on Pi 3 BT radio, and a malformed packed coming from the driver goes through the entire BT stack up to bthome-ble, giving you that incorrect measurement.
If you have an external BT dongle you can try to switch to that. Also, another possible workaround could be enabling encrypted Adv on sensor

pvvx, 2022-11-22:
Warning: If you have many BLE devices, the adapter must be able to receive BLE advertisements every 3 ms. UART speeds of 921600 baud and even USB1.1 are not enough to transfer information in binary form in the HCI format. Requires BT adapter with USB2.0 HS. Otherwise, there will be many gaps in the reception.

closed 2023-02-18 by pvvx