pvvx/ATC_MiThermometer #247: PIN code not working in ATC_v38.bin

Matush8, 2022-09-10:
Right after flash to v3.8 I set a new PIN code to 123456

In the log a have found expected messages:
PinCode: '123456'
Send pincode ok

Once disconnected there was no more connection to the device. My PC asks me for a PIN and when I enter 123456 there is a message "Connection failed"
I did this on brand new LYWSD03MMC B1.6

There was no problem with v3.7 and PIN.
I would like to do a downgrade however I'm unable to connect this device anymore because PIN code not working.

Any help?

Protoncek, 2022-09-10:
yep, been there... are you by any chance on Windows?
If so, try to connect with another PC, which you didn't connect to Xiaomi before. The problem (i guess) is that windows "kinda remembers" connection and even though it asks for password then in some strange way it doesn't use it. Try to find device in windows (it remembers it) and delete it. But, good luck if you're on win11... i didn't manage to find where these things are stored, at the end i connected with another PC and deleted password on xiaomi...

Matush8, 2022-09-11:
Yes, it's on the Win 10. I've tried another PC (also W10) and there was the same problem with "Connection failed".

Also tried another new LYWSD03MMC unit.
Just to be sure it's not related to that single one.

Now I have two thermometers with the same problem.

Protoncek, 2022-09-11:
Strange... are you sure that you never connected to devices with that "another PC"? try to connect with smartphone (or tablet).

In any way, i tried twice to enable this pin code, each time i've had so many problems and frustrations that i'll have to trash devices that i gave up with it. My 8 thermo's are without pin. I live in a house, so there's practically no chance that someone breaks into my thermometer... but, if it does...well...

pvvx, 2022-09-14:
https://github.com/pvvx/ATC_MiThermometer/issues/174#issuecomment-1003987084

Protoncek, 2022-09-15:
> Windows 10 with most BT adapters only supports bind and pin with one BLE device!

Jesus...really? I didn't know that.  That's a serious limitaton... what a bummer!

pvvx, 2022-09-15:
After binding to a BLE device, windows 10 reads and displays the battery level on the icon... Reads the Battery Service (0x180F)/Battery level state (0x2A1B) characteristic.
A situation is possible that shows last year's level or periodically makes a connection and reads the charge level. There were also such cases.

Matush8, 2022-09-16:
> [#174 (comment)](https://github.com/pvvx/ATC_MiThermometer/issues/174#issuecomment-1003987084)

Thank you, this helped a lot.

On my second PC I was able to connect with one of these two devices. However, I wasn't able to repeat this on my primary laptop.
I repeated pairing process with the same result until I did this:

Settings - Bluetooth & other devices > ATC_...... click on Remove device 
- It was older device and not the one I wanted to connect

After this I was able to pair with another thermometer and once OS asks for a PIN code it has to be entered without delay.

I can confirm this is not related to firmware v3.8 but to Windows OS. Unfortunately, it simply ignore pairing with another BLE device with PIN and still using stored PIN no matter you are connecting with another device.

Thank you all for you inputs

closed 2022-09-16 by Matush8