pvvx/ATC_MiThermometer #248: Request: Change number of digits

Mopfus, 2022-09-10:
Hello there. First of all, very nice work!
For me and my HomeAssistant Setup it would be really handy, if it would be possible to change the number of digits that are send. So choosing between 0.01, 0.1 and 1 would be great!
For my humidity, integers are fine. And for my temperature readings, 1 digit is enough.
Thank you in advance :-)

Protoncek, 2022-09-11:
Yes, i agree, that would be nice. However, i just created a separate sensors in HA and you're done:

```
- platform: template
   sensors:
    room_temperature_bthome:
      friendly_name: Room temperature
      unique_id: ani_temperatura_bthome
      value_template: "{{states('sensor.room_7d93_temperature')| round(1)}}"
      unit_of_measurement: "°C"
      device_class: temperature

    room_humidity_bthome:
      friendly_name: Room humidity
      unique_id: ani_vlaga_bthome
      value_template: "{{states('sensor.room_7d93_humidity')| round(0)}}"
      unit_of_measurement: "%"
      device_class: humidity
```

Mopfus, 2022-09-11:
Thats a good workaround. Thank you very much.

But as I have around 20 Sensors here, I would need 40 templates just to cancel out information, that is not really necessary or useful.

For me this many digits are  just "data-trash" that are send around where it is not necessary. They also suggest an accuracy of my $4 sensor that is far from being achieved.

pvvx, 2022-09-11:
The number of displayed digits is specified in the integration settings used in Home Assistant.
Example, description for integration with a [passive BLE monitor](https://github.com/custom-components/ble_monitor):
https://custom-components.github.io/ble_monitor/configuration_params#decimals

Mopfus, 2022-09-11:
> The number of displayed digits is specified in the integration settings used in Home Assistant. Example, description for integration with a [passive BLE monitor](https://github.com/custom-components/ble_monitor): https://custom-components.github.io/ble_monitor/configuration_params#decimals

Ah cool. I was not aware of that.
Sadly this is not possible with [BTHome](https://bthome.io) (which is available since HA 9.0). BTHome is per default on HA 9.0, which makes it super easy and flawless for using ATCs.

pvvx, 2022-09-11:
With limited accuracy and data coarsening, instead of a smooth normal graph, we get squares.
The sensors used in the thermometer give a relative accuracy better than 0.1% of full scale.
To more quickly determine the change - increase or decrease, even more accurate data is required than a difference of 0.01 from units of degrees or humidity. Used to maintain temperature and humidity. Especially with PID control.

For example, I am not satisfied with such temperature maintenance.
[image]

I wish there were even less fluctuations. But due to the limitation of 0.01 degrees and the inertia of the heater, it is impossible to get a more linear temperature in my warehouse in the country. Although the sensor parameters allow this.

pvvx, 2022-09-11:
> Sadly this is not possible with [BTHome](https://bthome.io) (which is available since HA 9.0). BTHome is per default on HA 9.0, which makes it super easy and flawless for using ATCs.

Displaying data in the required format is not solved by coarsening the data from the sensor.
Most programs use averaging of data over a period. With sensor data +20 +19 +20 the display total can be +19.66666... C

Mopfus, 2022-09-12:
Okay, thank you for this good explanations and thoughts.
I agree that "squares" for a temperature is not nice. For my use case they are not a problem.

Averaging over a period is of cause a good way to get more accurate values. And yes, then you could get a lot more digits.

Thats new to me, that BTHome is not working good. I just learned that it exists some days ago when they celebrated 9 years of HA (https://www.home-assistant.io/blog/2022/09/07/release-20229/). In the video they claimed, that it works well and you can use ESP32 as proxy to expand the range of your host-system. Sounds for me pretty neat and useful. I have more than 10 Sensors around and using BLE at the moment.  But I wanted to switch to BTHome. So maybe I should overthinkt that.

Protoncek, 2022-09-12:
> Thats new to me, that BTHome is not working good. I just learned that it exists some days ago when they celebrated 9 years of HA (https://www.home-assistant.io/blog/2022/09/07/release-20229/). In the video they claimed, that it works well and you can use ESP32 as proxy to expand the range of your host-system. Sounds for me pretty neat and useful. I have more than 10 Sensors around and using BLE at the moment. But I wanted to switch to BTHome. So maybe I should rethink that

Hm... i have 8 xiaomi's with pvvx FW via this new BTHome. I use one BT USB adapter and two ESP proxies. All works excellent. No drop-outs, quite ok temp readings, good signal strength. Before that i've had two ESP32 on ESPHome as BLE receivers - those also worked great, but this BLE proxy is even better - more flexible, since you don't have to decide which BT receiver will receive certain thermometer - system makes this decision based on signal strength. 
But, i don't request two digits precision, i take these readings more as "info" than laboratory measurement (which temp to two digits is, to be honest).

BTW...you can test new BThome system together with your old "system", if you have atc or pvvx FW on your xiaomi's. Only bad thing is that with atc or pvvx FW auto-discovery doesn't work in HA, you must add BTHome receiver (xiaomi) manually.

pvvx, 2022-09-14:
BTHome used to have a dual antenna solution, but it was built on a legacy ESP8266 module and BT via "AT".
And I don't see any new WiFi6 and BT5+ compatible implementations. But the modules have been around for a long time:
[image]

AythamiEsp, 2022-11-05:
So, what's the best way to change the digits to 0,1 accuracy? 
Or maybe the values could be rounded to 0,1x degrees C ; where x is 0 or 5. (to reduce squaring)

pvvx, 2022-11-05:
Value in 0.01.    1234 -> 12.34
Integer division:  [(Value_x01 + 5)/10 = Value_x1](https://github.com/pvvx/ATC_MiThermometer/blob/master/src/app.c#L272)

closed 2023-02-18 by pvvx