pvvx/ATC_MiThermometer #249: CGPR1

Seifon, 2022-09-12:
I hope support CGPR1，

Thanks

pvvx, 2022-09-15:
I've had several [CGPR1](https://pvvx.github.io/CGPR1/)s running for several months now and haven't found any required additions or changes.
They work in Qingping mode and fully meet my needs. Consumption in the official version is optimized enough for them.
What additional functionality can you suggest for CGPR1?

luizrrocha, 2022-09-15:
I am dying of curiosity about how to use the CGPR1 with this firmware. I could not find any meaningful explanation on how to interpret the sensor´s messages, just your info on the messages themselves...

Did you find or created any tutorial on that?

Thanks,

Luiz

Seifon, 2022-09-15:
> I've had several [CGPR1](https://pvvx.github.io/CGPR1/)s running for several months now and haven't found any required additions or changes. They work in Qingping mode and fully meet my needs. [...]

I hope the author can produce a .bin firmware for the VGPR1 human motion sensor. Because I want to support BTHome through this third-party firmware and send motion sensing Bluetooth signals

我希望作者能为VGPR1 人体动作传感器 出一个固件，想通过这个三方固件支持BTHome，发送动作感应蓝牙信号

Seifon, 2022-09-15:
> I am dying of curiosity about how to use the CGPR1 with this firmware. I could not find any meaningful explanation on how to interpret the sensor´s messages, just your info on the messages themselves...
> 
> Did you find or created any tutorial on that?
> 
> Thanks,
> 
> Luiz

I hope the author can produce a .bin firmware for the VGPR1 human motion sensor. Because I want to support BTHome through this third-party firmware and send motion sensing Bluetooth signals

我希望作者能为VGPR1 人体动作传感器 出一个.bin固件，想通过这个三方固件支持BTHome，发送动作感应蓝牙信号

pvvx, 2022-09-19:
Connect CGPR1: 
[image]
https://github.com/pvvx/pvvx.github.io/blob/master/ATC_MiThermometer/TelinkMiFlasher.html#L570

Qingping / ClearGrass  Advertisement:
https://github.com/pvvx/pvvx.github.io/blob/master/ATC_MiThermometer/TelinkMiFlasher.html#L402

Sorien, 2022-09-22:
maybe there is one security use case, bthome allows defining own encryption key for advertisement so it can't be misused by someone to find out if somebody is inside the house.

pvvx, 2022-09-23:
You can install the binkey encryption key in TelinkMiFlasher.
The sensor will switch to MiHome ([Mijia](https://github.com/pvvx/ATC_MiThermometer/tree/master/InfoMijiaBLE)) mode.

closed 2023-12-23 by pvvx