pvvx/ATC_MiThermometer #252: Bluetooth announce the 5 entities

Blackshome, 2022-09-27:
Problem: We are having fast battery usage and unnecessary data into HA (Home Assistant). BTHome is the fastest to drain the battery life (like 1 to 2 weeks down to 73%) and MIJIA is not bad but could be better. We are on V3.8 using a Xiaomi Mijia LYWSD03MMC. Every time a sensor updated a value like 00.01 of a degree it sends a update to HA even if you set the "advertising interval" & the "Measure Interval". It just drains the battery life.

Solution: Have a options to announce the 5 entities v Bluetooth (yes or no / or tick box) and then have the decimal option in a drop down list like (00, 00.0, 00.00, 00.000) or a text box to enter one of the 4 values (like 00.0) next to each entity (we don't need 13 Humidity & voltage sensors around our house just the temp & battery level). This would resolve everything allowing the user to select what entity will be broadcasted v Bluetooth & what value they need (a temp value of 20.23 can just be 20.2 or just 20, user decides and when it changes to 20.3 or 21 it then send the value v Bluetooth). This will increase the life of battery power not to mention unnecessary data flying around everywhere. Do everything at the sensor not in HA or any other platform you use.

Thank you for all the hard work you have done to date & hope you consider this change.

pvvx, 2022-09-28:
What is draining the battery? Number of digits? :)
BLE is not Zigbee.
BLE advertising is transmitted with a period of up to 10 seconds - a limitation of the standard. Most operating systems start to misbehave if BLE advertisements are transmitted with a period of more than 2.5 seconds. Connection may not be possible.

BLE ads are transmitted without acknowledgment, and if you have a lot of devices and WiFi is still working, then multiple losses (collisions and interference) occur.
Those. the device must always send BLE advertisements, with a period of no more than 2.5 seconds. Otherwise, nothing will work for half of the users.

> BTHome is the fastest to drain the battery life (like 1 to 2 weeks down to 73%)

Active scanning increases battery consumption by many percent. The BLE device responds to active requests, which consumes additional power.

The CR2032 battery voltage is read at load current 1..3 мА.
At the same time, the CR2032 - the first two weeks the battery gives off more than 3 volts, and then the nominal level is set - about 2.8..2.9V. And so it goes on for several months.
The consumption is pulsed and if the recommended capacitors for power are installed (which are not in), then the picture changes - the battery voltage can last more than 3V for a very long time. 
On a new battery and more than 3.3V.
The formula for converting battery voltage to % is used: 
battery_level = (battery_mv - 2200) / ((3100 - 2200) / 100)

(2850-2200)/(3100-2200)/100 = 72.2% - This is the stable voltage of a new CR2032 battery (+25С).
The formula is not mine, but is used in external software and when changed, there will be inadequate readings.

PS: Everything listed here has already been described many times and please study these issues on your own, and not invent unsolvable proposals.

pvvx, 2022-09-28:
> Problem: We are having fast battery usage and unnecessary data into HA (Home Assistant).

Solution: In the list of HA objects, deactivate unnecessary ones - check the box and save.

> drop down list like (00, 00.0, 00.00, 00.000)

Solution: In HA, in the integration BT settings for each device, select the number of decimal places to display. And an update period.

Blackshome, 2022-10-01:
Hi @pvvx 

Thanks for all the info.....

> Solution: In the list of HA objects, deactivate unnecessary ones - check the box and save.

I have deactivated them but on the HA forum it was suggested that even doing that the data would be sent anyway by the sensor thus still using battery and draining it. Not sure if that is the case but it made sense to me as it is about the sensor (device) sending data using battery not HA receiving the data. But I am not a expert on Bluetooth and happy to learn the truth from your above post :)

> Solution: In HA, in the integration BT settings for each device, select the number of decimal places to display. And an update period.

I cant see were that option is available. Can't see it in the device or entity. No yaml for sensors as they are auto discovered. If you could point me in the right direction I would be love to change the settings, happy to learn and thank you.

> Active scanning increases battery consumption by many percent.

My understanding is the new Bluetooth proxy is passive, but will keep that in mind.

> CR2032

All your information on the CR2032 is very interesting, so thank you :)

> PS: Everything listed here has already been described many times and please study these issues on your own, and not invent unsolvable proposals.

Sorry, but thank you for taking the time and respond to my post. 

> Your suggestions only increase the size of the firmware and consumption, but do not give any real benefit. External software displays the average value from several measurements and the number of displayed digits is set in it.

Ok, thank you

pvvx, 2022-10-10:
[Configure](https://custom-components.github.io/ble_monitor/configuration_params) [Bluetooth Low Energy Monitor](https://github.com/custom-components/ble_monitor) (ble_monitor)
[image]

[image]

pvvx, 2022-10-10:
>BTHome быстрее всего разряжает батарею (например, от 1 до 2 недель до 73%),

CR2032 battery life test in LYWSD03MMC HW:B1.9 (default settings) outdoor:

[image]

10 months, starting from -25C at the beginning of December 2021. On October 2, 2022, when the temperature dropped to +2C, CR2032 could not output more than 2 Volts under load. At +25C another thermometer is powered by this battery.
The graph clearly demonstrates that there is no way to determine the discharge level of CR2032 by voltage...

Blackshome, 2022-10-22:
Hi @pvvx 

Thanks for the information. We don't use that custom component for HA. They also have a **Important announcement about the future of BLE monitor** and will probably be deprecated. Hope HA will make the option so we can select how many decimal points to display.

I still think it would be really nice to have a option in your firmware to select a yes or no option for BLE ads are transmitted for each of the 5 entities and then have the decimal option in a drop down list like (00, 00.0, 00.00, 00.000) or a text box to enter one of the 4 values (like 00.0) next to each entity.

[image]

I have moved all my sensor from BTHome back to MIJIA for the Advertising type as I prefer that format (temp = 21.5) rather than BTHome (temp = 21.53). Using MIJIA advertising type, the battery seems to be ok again but maybe I should test BTHome and see if that drains it fast again. I am not saying that BTHome drains the battery fast as I would need to test it again now having a understanding of the CR2032 but I just don't like the format and I can't change it unless I use the custom component you suggested above that is going to be deprecated. Ernst79 advise is, when all your sensors are available in Home Assistant, make the move. 

I have 3 LYWSD03MMC using MIJIA for the Advertising type and after 12 months the battery are 43%, 60% & 61%.... Not bad :)

closed 2023-02-18 by pvvx