pvvx/ATC_MiThermometer #253: Unable to view/change settings after flashing ATC v3.8

Tediore, 2022-10-08:
Did an OTA update to ATC v3.8 on two of my three Xiaomi thermometers (previous firmware v2.4), and I'm now unable to view/change settings in the web flasher. The devices connect, but the web flasher says "detected custom firmware" and the settings don't load. When I tap "get settings" it says "error on sending setting 55." It seems to disconnect shortly afterward too. I tried changing the batteries in the v3.8 sensors, but that didn't work. The thermometers appear to be functioning well otherwise.

I flashed the third one with ATC v3.7, and I'm able to view/change the settings without any issues, so maybe there's an issue with the web flasher and v3.8? Please let me know if there's anything else I can try.

Tediore, 2022-10-08:
Well, after many other attempts, I was able to connect to the two sensors. Not sure what the issue was, but we're good now!

closed 2022-10-08 by Tediore

Protoncek, 2022-10-08:
I have 8 of these devices and, yes, sometimes multiple tries are needed in order to connect. Being very close doesn’t help.  Misterious…

Tediore, 2022-10-08:
Could just be RF interference. This firmware is so cool though!