pvvx/ATC_MiThermometer #255: Xiaomi Mijia - setting and clearing GPIO PA5 via bluetooth connection - detailed steps

flo-x, 2022-10-09:
I would like to connect to a Xiaomi Mijia from an esp32 and set or clear GPIO PA5, just as is shown in README.md to command a LED. Can someone tell me the steps I need to take? My understanding is that I need my esp32 to connect to the Mijia, then write to some characteristic:

> "Characteristic UUID [0x1F1F](https://github.com/pvvx/ATC_MiThermometer#primary-service-uuid-0x1f10-characteristic-uuid-0x1f1f) - Notify, frame id 0x33 (configuring or making a request): temperature x0.01C, humidity x0.01%, battery charge level 0..100%, battery voltage in mV, GPIO-pin flags and triggers."

What exactly should I write to set or clear PA5?

Thank you!

pvvx, 2022-10-10:
Custom Main service (0x1f10) - '00001f10-0000-1000-8000-00805f9b34fb'
Custom read-write characteristic (0x1f1f) - '00001f1f-0000-1000-8000-00805f9b34fb'

Output GPIO_PA5 work only TRG off! - If Temperature hysteresis = 0 and Humidity hysteresis = 0

Set pin to "1", GPIO_PA5 PullUp 10 kOm: sendCustom 4501 (2 bytes 0x45,0x01)
Set pin to "0", GPIO_PA5 PullDown 100 kOm: sendCustom 4500 (2 bytes 0x45,0x00)

flo-x, 2022-10-14:
Thanks a lot, works perfectly!

closed 2022-10-14 by flo-x